gpiozero/gpiozero #578: Can we have more than 1 mockpin?

cvanlabe, 2017-07-17:
In my code unittests I'm running on a non-pi I need more than 1 pin.
The workaround I have is to 'dunder __del__ () ' the mocked gpiopin, so it can be reused,  but that simply doesn't work in code where multiple need to be initialized.

Can we have support for more than 1 pin please? If there is such a way already, could you help me understand how to do it?

waveform80, 2017-07-17:
I'd suggest having a look at the existing test suite. The interesting bits are as follows:

* The [teardown_function](https://github.com/RPi-Distro/python-gpiozero/blob/master/tests/test_inputs.py#L19) (run after each test function) calls `MockFactory.reset` which clears down the pin cache so the next time the pin is constructed it's a distinct object.

* The [test functions themselves](https://github.com/RPi-Distro/python-gpiozero/blob/master/tests/test_inputs.py#L67) which first call `MockFactory.pin` to get (for example) the GPIO4 pin instance, then construct a device on GPIO4. Due to `Factory.pin` being idempotent the device will get the exact same pin instance which allows the mock pin to affect the state of the device under test.

* [Some tests use derivatives](https://github.com/RPi-Distro/python-gpiozero/blob/master/tests/test_inputs.py#L162) of `MockPin` with special properties (like pretending they're attached to a capacitor which delays the time it takes the pin to raise)

Ultimately the [`MockFactory`](https://github.com/RPi-Distro/python-gpiozero/blob/master/gpiozero/pins/mock.py#L412) is a fairly simple affair and happily provides more than one pin for testing, but like all other factories if you ask for the same pin number multiple times you'll get the same object (after all the instances you request all represent the same physical chunk of hardware).

cvanlabe, 2017-07-17:
Thanks! That looks good.. I'll have a look at that and circle back if I would have any issues.

bennuttall, 2017-07-27:
Closing - feel free to comment if you have any further questions

closed 2017-07-27 by bennuttall