gpiozero/gpiozero #580: Cannot initialize DigitalOutputDevice with active_high=False

sumitkm, 2017-07-24:
Hi,
I am trying to drive relays through my Pi and can't seem to initialize them so that they don't turn ON (switch to NC, default is NO). My code is as follows.

I am trying to Initialize the relay so it maintains its default state (OFF, in NO). Then based on user input turn it on or off.

With the following code the relays turn on as soon as they are initialized. Ignore on() or off() requests and go off only when closed.

I have tried all combinations for active_high (True/False) and initial_value (True/False). The behavior is the same for all cases.

```

from gpiozero import DigitalOutputDevice
from time import sleep

from uuid import getnode as get_mac
mac = get_mac()

print(mac)

relay1 = DigitalOutputDevice(pin=17, active_high=False, initial_value=True)
relay2 = DigitalOutputDevice(pin=27, active_high=False, initial_value=True)

print("initialized")
action = input("Enter 1 to turn on, 0 to turn off: ")

while action !=9:
        if action == 1:
                relay1.on()
                relay2.on()
                print("on")
        elif action == 0:
                relay1.off()
                relay2.off()
                print("off")
        action = input("Enter 1 to turn on, 0 to turn off: ")

relay1.close()
relay2.close()
print("Bye")
```

Any inputs much appreciated. I am a bit of a Python noob so please pardon any obvious boo boos.

P.S. The PR tied to issue #66 refers to a Relay class but I couldn't initialize a Relay looks like it has been taken out again. Any particular reason? Its fine to have it undocumented ;-)

bennuttall, 2017-07-25:
Your code doesn't look wrong. I would suggest you try as little code as possible to determine whether it's really not working as expected. Try opening a Python shell (not a file) and trying this:

```python
$~ python3
>>> from gpiozero import *
>>> relay = DigitalOutputDevice(17, active_high=False)
>>> relay.on()
>>> relay.off()
```

and see if that works as expected. If it does, then it's just down to the logic in your longer script turning it off. If not, perhaps there's something about your relay that works differently. Do you have a datasheet? Or do you have any working code using another library? If so, please post these and we'll know more.

> P.S. The PR tied to issue #66 refers to a Relay class but I couldn't initialize a Relay looks like it has been taken out again. Any particular reason? Its fine to have it undocumented ;-)

[This comment](https://github.com/RPi-Distro/python-gpiozero/issues/66#issuecomment-148020962) in #66 explains why we removed `Relay`.

sumitkm, 2017-07-25:
Thanks @bennuttall ,

I tried the script from shell. The relay turns on as soon as the constructor is called. Doesn't wait for on and as with the bigger script, doesn't respond to on/off.

I think I am missing something obvious. 
I'll go back to basics and see if I got any fundamentals wrong with the usage. 
I will also try the C# library I wrote once upon a time. That one is pretty barebones. 

Will update back here.

And, thanks, I missed that comment about removing the Relay class, in the thread... Late night effect 😊 

Cheers

sumitkm, 2017-07-25:
Okay, I tried RPi.GPIO and got the exact same result.

```
>>> import RPi.GPIO as GPIO
>>> GPIO.setmode(GPIO.BCM)
>>> GPIO.setup(17, GPIO.OUT, initial=GPIO.HIGH)
>>> GPIO.cleanup()
```

As soon as I call GPIO.setup, irrespective of initial value (GPIO.HIGH/GPIO.LOW), the relay gets triggered.

As with gpiozero calling GPIO.output(...) doesn't change the state.

So its clear the Relay is triggered as soon as the PIN is setup (too easily?). Anyone has any clues why? It's a cheap relay module, I'll see if I can find the data-sheet for it.

It is clearly not a library bug so closing this issue. Will report back if I get to the bottom of this!

Cheers.

closed 2017-07-25 by sumitkm

jdimpson, 2018-08-18:
This looks like it's the same cause as #613, which at its root is a bug in RPI.GPIO (see [https://sourceforge.net/p/raspberry-gpio-python/tickets/145/](https://sourceforge.net/p/raspberry-gpio-python/tickets/145/). The fix is marked to be included in the next release (0.6.4) but not sure when that will be.