gpiozero/gpiozero #582: Upgrade to gpiozero 1.4 on Jessie breaks gpiozero

craignied, 2017-07-30:
Sorry to cross post this from StackExchange, but I figured this would be useful here, too:

On a stock Jessie install, gpiozero works. E.g.

    >>> from gpiozero.devices import pin_factory
    >>> print(pin_factory)
    <class 'gpiozero.pins.rpigpio.RPiGPIOPin'>

I then upgrade to gpiozero 1.4 with

    sudo apt update
    sudo apt full-upgrade

Which I'd like because the pinout command is super useful, and gpiozero is among the upgrade:

    The following packages will be upgraded:
      bind9-host debconf debconf-i18n debconf-utils gtk2-engines-pixbuf
      libbind9-90 libc-ares2 libdns-export100 libdns100 libdvdnav4 libgcrypt20
      libgnutls-deb0-28 libgnutls-openssl27 libgtk2.0-0 libgtk2.0-bin
      libgtk2.0-common libirs-export91 libisc-export95 libisc95 libisccc90
      libisccfg-export90 libisccfg90 liblwres90 libsmbclient libtiff5 libwbclient0
      perl perl-base perl-modules python-gpiozero python3-gpiozero python3-thonny
      rpi-chromium-mods samba-common samba-libs xarchiver

But although pinout works (it's so cool!) nothing else does, e.g.

    >>> from gpiozero.devices import pin_factory
    Traceback (most recent call last):
      File "<stdin>", line 1, in <module>
    ImportError: cannot import name pin_factory

And none of my python code with

    from gpiozero import ...

Which used to work does either. Any ideas? Thanks in advance!

waveform80, 2017-07-31:
I'm afraid `pin_factory` moves in 1.4 to become a class attribute of `Device`. In other words, import `Device` from `gpiozero` and you can then access the current pin factory as `Device.pin_factory`. For example:

```python
from gpiozero import Device

a_pin = Device.pin_factory.pin(2)
```

However, there's rarely a need to do this now as the other major change is that instead of devices accepting either a pin specification (e.g. 2) or a Pin instance, they only accept a pin specification, but have the ability to accept an overridden `pin_factory`. In other words, if previously you did something like this:

```python
device = SomeDevice(my_factory.pin(2))
```

It now becomes:

```python
device = SomeDevice(2, pin_factory=my_factory)
```

It is a [breaking change](http://gpiozero.readthedocs.io/en/stable/changelog.html#release-1-4-0-2017-07-26) (for which I take the blame), but I hope I can justify it. The major reasons behind the change are as follows:

1. It was impossible under the old system to specify different pin factories for boards with fixed pin specifications (e.g. RyanteckRobot). One could do so for a single board by changing the default pin factory, but changing factories implicitly closed the old one so it was impossible to have, e.g. two remote separate remote factories for two boards. Incidentally, switching factories no longer closes the old factory in 1.4.
2. There were some tricky object life-time issues associated with having the pin factory as a global (particularly when multiple pin factories were instantiated; these weren't obvious in 1.3 but came up during 1.4 development when the pin reservation system was introduced to support sharing of pins for things like SPI).

I can only apologize for not getting the design right first time, but API design is a trickier thing than many imagine. Unfortunately there was no easy way to support both APIs in 1.4, and given the pins API was marked unstable (in 1.3.x) I felt we could get away with a breaking change (that warning is removed in 1.4; the pins API is now committed to being stable).

You can find lots of examples of playing around with pin factories in the new [remote GPIO recipes chapter](http://gpiozero.readthedocs.io/en/stable/recipes_remote_gpio.html) of the docs.

bennuttall, 2017-07-31:
> And none of my python code with `from gpiozero import ...`

None? Does `import gpiozero` work?

As Dave said, some advanced functionality changed between 1.3 and 1.4 but we'd been relatively quiet about publicising those features until now (as they're now stable).

bennuttall, 2017-07-31:
I've just run an upgrade and `import gpiozero` works fine. I guess it comes down to some pin factory stuff that got changed. Good to hear you've been using it - sorry for the bc break, but I hope you find the new syntax easier to work with.

craignied, 2017-07-31:
Ah, I read that section and didn't quite get it, now I do :) So this works now for sure:

> >>> from gpiozero import Device
> >>> print( Device.pin_factory )
> <gpiozero.pins.rpigpio.RPiGPIOFactory object at 0x765d6c50>

But this, which worked before, still doesn't. I *think* I still use integers for pins. What happens now is that it sits for a while and then outputs 1.0 regardless of what's in front of the HC-SRO4. Any ideas?

> from gpiozero import DistanceSensor
> ultrasonic = DistanceSensor(echo=17, trigger=4)
> ultrasonic.distance

bennuttall, 2017-07-31:
That will be related to #385 and #435. I *think* it's buggy in Python 3 but not Python 2, unfortunately. Nothing specific to `DistanceSensor` changed in v1.4.

FYI you're using `>` (blockquote) for your code. You probably want to use backticks (```) instead. Single backticks for inline code, 3 backticks (optionally followed by the word `python`) for a nested code block.

closed 2017-07-31 by bennuttall