gpiozero/gpiozero #583: Neopixel support

bennuttall, 2017-08-04:
A big task, but we should see if we can add neopixel support.

MauiJerry, 2017-10-23:
check the pi neopixel library (https://github.com/jgarff/rpi_ws281x) for example code. 
since that is all built separately from rest of this system, it might take a fair bit of effort.

jantman, 2023-07-20:
I'd posit that these days, a library without Neopixel support will probably be ignored (with frustration) by a *lot* of people...

bennuttall, 2023-07-21:
Don't write comments like that. Open source maintainers don't owe you anything.

gpiozero had simple goals to make it easier for educators to teach basic physcial computing. We expanded the library a lot but things like I2C and neopixels add a whole layer of complexity to it. There are other libraries you can use to control neopixels and you can use them alongside gpiozero.

dsssssssss9, 2023-07-21:
> I'd posit that these days, a library without Neopixel support will probably be ignored (with frustration) by a _lot_ of people...

Not a very kind thing to say!

You can  successfully run GPIZERO & separate Neopixel / WS2812 library simultaneously 

Perhaps you should try this ?