gpiozero/gpiozero #584: Ultrasonic sensors always return max_distance since update

Exomene, 2017-08-06:
Hi,

Since the last update my ultrasonic sensor always return their max_distance and take far more time than before to initialize.

After checking the connections on the breadboards and changing the sensors for brand new one the problem still persists. The script at https://bitbucket.org/MattHawkinsUK/rpispy-misc/raw/master/python/ultrasonic_2.py not using gpiozero returns coherent values on the very same breadboard and sensors. And to be sure it was not a coding issue I used the script given in the docs at http://gpiozero.readthedocs.io/en/stable/recipes.html#distance-sensor which still return the max_distance.
I also uninstalled the library installed via apt and installed it via pip and the results are still the same.

Can you show me a way to solve this problem?

Best regards

Exomene, 2017-08-07:
Reverting to 1.3.2 and 1.3.1 via pip didn't change the behaviour.

bennuttall, 2017-08-07:
Unfortunately this is a known bug - we'll try to get it fixed for v1.5.

Closing as this is a duplicate of #385 and #435.

Exomene, 2017-08-08:
Great!
I'm looking forward to that new version

purbeckstem, 2017-12-02:
I've spent 3 days of tinkering trying to get gpiozero working with my HC-SR04 for a robot workshop at school, failed repeatedly, then checking it with arduino, RPi.gpio and come to the conclusion that it's broken.  I've now managed to find this in github, so I'm not mad or inept.   

 Since it is broken, please could the readthedocs.io documentation be updated to show there is a bug to stop people investing time in using that implementation, and just writing their own?

getreup, 2017-12-11:
I have the same issue. Is there a timeline on this?

waveform80, 2018-02-16:
I've dug into this today; this issue shouldn't really have been closed as it's not an exact duplicate of #385 and #435 which described something slightly different (subtle inaccuracy rather than gross inaccuracy with a large initialization delay). Anyway, I'll re-open this for the sake of tagging the PR correctly as I've pushed one (#617) that should fix *almost* everything to do with the DistanceSensor, except for remote operation over pigpio (which is going to require more work in 1.5, specifically down in the pin implementations).

bennuttall, 2018-02-19:
@waveform80 can this be closed now?

waveform80, 2018-02-19:
Merged in #617

closed 2018-02-19 by waveform80

asenturk22, 2022-01-12:
Hi Dave,  I have a problem my UDS,    can you help me?

bennuttall, 2022-01-12:
A 4 year old issue is not the right place to ask for help. Please use one of the recommended support channels: https://gpiozero.readthedocs.io/en/stable/#issues-and-questions