gpiozero/gpiozero #585: sample_rate doesn't work for LineSensor

wpballa, 2017-08-06:
sensor=LineSensor(27)
sensor.threshold=0.25
sensor.sample_rate=1.0
fails on the last line as a nonexistent attribute. Comment it out and everything else works.

As a second possible issue, I think the active and inactive are switched. The sensor line is high (about half of the 3.3V) and the when_no_line() is activated. Rather counterintuitive.

bennuttall, 2017-08-07:
There is no `sample_rate` attribute. It's an init parameter only, and can't be changed after instantiation. The correct usage is:

```python
sensor = LineSensor(27, threshold=0.25, sample_rate=1.0)
```

Hope that helps!

closed 2017-08-07 by bennuttall

bennuttall, 2017-08-07:
Interestingly, without our over-the-top warnings, you wouldn't have known anything was wrong. Usually, Python allows you to set arbitrary attributes on objects, for example:

```python
class Thing:
    def __init__(self):
        self.name = "a thing has no name"

    def speak(self):
         print('My name is {}'.format(self.name))
```

```python
>>> thing = Thing()
>>> thing.thing_name = "Bob"
>>> thing.speak()
My name is a thing has no name
```

Here, the user has incorrectly set the attribute `thing_name`, and that was ok (no error), but it didn't do what the user expected it to.

In GPIO Zero, we prevent users from setting attributes which weren't set at the point of instantiation, and instead give a warning, because it's more likely they made a typo or got the name wrong than actually intended to set an unused attribute, e.g:

```python
button.pressed = foo
```

This would usually set an unused attribute `pressed` to the function `foo`. But that doesn't do what the user intended to do (to set up a callback to fire when the button is pressed), because correct usage is:

```python
button.when_pressed = foo
```

wpballa, 2017-08-18:
Ben, Just a suggestion, but it isn't clear in the documentation which attributes are available only at instantiantion time vs those that can be changed later. However, everything is working great now that I understand the call procedure.