gpiozero/gpiozero #587: GPIOzero using LED, LightSensor, and MotionSensor

yimiblu, 2017-08-18:
Hi all!
I'm having troubles trying to make my project. 
Maybe it's all my own fault, because of my short Python experience :D.

The main thing is that i need to concatenate 3 actions, and i can't find the right way!

I have a 'led', it's working ok.
I have a motion sensor (PIR): it works ok.
I have a LightSensor (LDR): it works ok.

They all work fine separately, but i need a hand with the code to make them all work simultaneously:

thing1 = lightSensor determines if it's light or dark
thing2 = motionSensor determines if there is any movement
thing3 = led is triggered on or off

if (thing2 = movement) AND (thing1 = dark), then thing3 is triggered ON
if (thing2 = movement) AND (thing1 = light), then thing3 is triggered OFF

This is the esential behaviour i need. I have tried some solutions read here, but i think that i can't manage the three things the right way. Any kind help from you masters?

I now it must be a loop i can't imagine :( 'while motion is not detected: do nothing' but when motion is detected, measure the light, and activate the led if it's dark. And while motion is detected, continue light metering to ensure led is ON or OFF until motion ends, and so on.

Thanks! (and please, excuse my english is not as good as i'd need :D , i'm from Spain)
Javi

ukBaz, 2017-08-18:
Hi Javi,

I have no way of testing this as I don't have the hardware... Here is a suggestion of how to structure the code for this task:

```python
from signal import pause
from gpiozero import MotionSensor
from gpiozero import LightSensor
from gpiozero import LED

pir = MotionSensor(4)
ldr = LightSensor(18)
led = LED(17)

def on_motion_detected():
    if ldr.light_detected:
        led.on()
    else:
        led.off()

if __name__ == '__main__':
    pir.when_motion = on_motion_detected
    pause()
```

I hope that helps.

Regards,
Barry

bennuttall, 2017-08-18:
I'm not 100% sure of your requirements. You said:

> if (thing2 = movement) AND (thing1 = dark), then thing3 is triggered ON
> if (thing2 = movement) AND (thing1 = light), then thing3 is triggered OFF

But I think you mean:

- If movement and darkness => led on
- Otherwise => led off

i.e the LED should only be on if there is both movement and darkness. If only one of these, or neither, the LED should be off.

There are a few different approaches you could take. This approach is quite easy to follow, and just keeps checking the state of both sensors in a loop:

```python
from gpiozero import MotionSensor, LightSensor, LED

pir = MotionSensor(4)
ldr = LightSensor(18)
led = LED(17)

while True:
    if pir.motion_detected and not ldr.light_detected:
        led.on()
    else:
        led.off()
```

Barry's approach above fires an event when motion is detected and checks whether light is detected - though his light/dark logic is wrong, should be:

```python
def on_motion_detected():
    if ldr.light_detected:
        led.off()
    else:
        led.on()
```

A third approach is more declarative, where the state of the LED is specified in one line:

```python
from signal import pause
from gpiozero import MotionSensor, LightSensor, LED
from gpiozero.tools import all_values, negated

pir = MotionSensor(4)
ldr = LightSensor(18)
led = LED(17)

led.source = all_values(pir.values, negated(ldr.values))

pause()
```

Setting a device's source means you can tell it where its values from:

- `pir.values` is the stream of True/False values from the motion sensor
- `ldr.values`is the stream of True/False values from the light sensor
- `negated(ldr.values)` is a stream of the opposite values from the light sensor
- `all_values` takes the two sensor values and yields a stream of True/False values using the `all` function, i.e. True if all (both) of the values are True

Hope that helps!

closed 2017-08-21 by bennuttall

yimiblu, 2017-08-21:
Absolutely grateful for your help! I'll study both answers with all stuff connected. I promise results, and and posterior analysis of what was wrong, and what worked! 

In a more detailed approach, i can tell that what i called LED, is actually a RELAY, with manage an external circuit where there is a bunch of IR LEDs and a power source. But, as the relay contains an optocoupler, i assume its behaviour as a LED. Maybe i'm wrong? 

Thanks again, there we go!
Javi

yimiblu, 2017-08-21:
Hi again I'm not sure if this thread has been closed... :/

I'd like to share more info about this question :)

I've tried both ideas.

First one: i haven¡t managed to get it working, maybe por this part:

> if __name__ == '__main__':
    pir.when_motion = on_motion_detected
    pause()

I'm not sure if i must adapt it or if it's literal :(

Regarding second solution, it seems to work quite fine! At least, it does what it says: led turns on when movement occurs in low light conditions. 👍 👍 

Now some thoughts:

- I dont know that determines if light is low or bright enough to activate the LED. I assume that capacitor values of the LDR is being read, and library accepts some value as the Threshold. I wonder if there's an easy way to set that Threshold value.
- It could be similar with MotionSensor i guess. But in this case, as sensor has two potentiometers, even when you set time as minimun, and sensitivity to maximun, there exists a period of time where detection is not working. Is there something to do with this behaviour using gpiozero library?
- In both cases, maybe it's possible to manage more precission using RCTime with GPIO library... ¿?

Thank you again for your fast, generous and useful help.

Javi

yimiblu, 2017-08-21:
Ups!! I forgot to mention that both solutions provided by @bennuttall works apparently the same, but first one seems to be easiest for me to add the rest of my code pieces. ^_^

bennuttall, 2017-09-04:
> Hi again I'm not sure if this thread has been closed... :/

Sorry for the confusion (and lack of response - I've been away for 2 weeks). The issue has been closed (as we have provided a solution to your problem), but you can still comment.  I just didn't see the comments as I was on holiday.

> In a more detailed approach, i can tell that what i called LED, is actually a RELAY

Ok, that's fine. But I would use the generic `OutputDevice` class instead of `LED`. In my experience, relays tend to have the opposite logic to LEDs, i.e. the GPIO pin going low turns it on, high turns it on. The way you do this is with `relay = OutputDevice(17, active_high=False)`.

> I'm not sure if i must adapt it or if it's literal :(

That's literal. It's not strictly necessary but it's a way of ensuring a bit of code only executes if the script is run directly (rather than imported from another file). See https://stackoverflow.com/questions/419163/what-does-if-name-main-do

> I dont know that determines if light is low or bright enough to activate the LED. I assume that capacitor values of the LDR is being read, and library accepts some value as the Threshold. I wonder if there's an easy way to set that Threshold value.

With all of these components you're using, it's probably best to test them out by themselves before connecting everything together in one program. The light sensor does indeed have a light/dark threshold. The default is `0.1` (so a small amount of light is considered as "light" not "dark"), but it can be changed. See the parameters in the [documentation](https://gpiozero.readthedocs.io/en/stable/api_input.html#light-sensor-ldr).

Try a loop looking at the light values with the sensor on its own:

```python
from gpiozero import LightSensor

sensor = LightSensor(4)

while True:
    print(sensor.value)
```

> It could be similar with MotionSensor i guess. But in this case, as sensor has two potentiometers, even when you set time as minimun, and sensitivity to maximun, there exists a period of time where detection is not working. Is there something to do with this behaviour using gpiozero library?

`MotionSensor` is smoothed out, light `LightSensor` is. See the [source code](https://github.com/RPi-Distro/python-gpiozero/blob/master/gpiozero/input_devices.py#L381). The default threshold for a motion sensor is `0.5`, but again, can be [changed](https://gpiozero.readthedocs.io/en/stable/api_input.html#motion-sensor-d-sun-pir).

> In both cases, maybe it's possible to manage more precission using RCTime with GPIO library... ¿?

I'm not sure!

> Ups!! I forgot to mention that both solutions provided by @bennuttall works apparently the same, but first one seems to be easiest for me to add the rest of my code pieces. ^_^

That's true. GPIO Zero is a multi-paradigm API, meaning you have different ways of achieving the same thing. Some methods are more suitable in some situations over others. Procedural code is always the easiest to start with - the declarative approach is more advanced, more complex to understand, but it's very powerful what you can do with very little code. See [this article](https://www.raspberrypi.org/blog/gpio-zero-update/).