gpiozero/gpiozero #589: Booleanized .values?

bennuttall, 2017-08-21:
I realised while answering #587 that while some devices have linear value scales, they also present a boolean state, for example the light sensor `.value` might be `0.5` but its `.light_detected` is `True`.

In the procedural code solution to the problem, the light sensor value was unused, and just the boolean state was used:

```python
while True:
    if pir.motion_detected and not ldr.light_detected:
        led.on()
    else:
        led.off()
```

However, when using source/values, the `.values` property yields float values, not booleans, so I had to manually booleanise the `.value` using the same logic that's found in `light_detected` (`is_active`):

```python
led.source = all_values(pir.values, booleanized(ldr.values, 0, ldr.threshold))
```

So I was thinking, it would be much neater if we provided a `.values`-like iterator for the boolean state, for example:

```python
@property
def bool_values(self):
    while True:
        yield self.is_active
```

Light and dark on their own would become:

```python
ldr.bool_values  # light
negated(ldr.bool_values)  # dark
```

Making my solution neater:

```python
led.source = all_values(pir.values, negated(ldr.bool_values))
```

Like many problems, this is easy to solve, but the hard bit is *naming it*. Any ideas? @waveform80 @lurch @martinohanlon

lurch, 2017-08-21:
```python
ldr_bool_values = booleanized(ldr.values, 0, ldr.threshold)
led.source = all_values(pir.values, negated(ldr_bool_values))
```
is quite close to your last example? :wink: 

But yeah, for those devices which have an 'implicit' boolean state like `light_detected` it _might_ be nice to offer a simpler way to use that with source-values.
Instead of naming it `bool_values` what about `active_values`? So that `values` returns an infinite generator of `value`, and `active_values` returns an infinite generator of `is_active`?  Perhaps the LDR class could even then alias `lights` to `active_values`? But again, this becomes a tricky naming problem... :confused: 
```python
led.source = all_values(pir.values, negated(ldr.lights))
```

Of course the advantage of explicitly using `booleanized` is that it makes it easy to use the funky hysteresis option I added, to prevent 'flickering' when the LDR is hovering around the threshold value.

bennuttall, 2019-03-31:
Ok, just a thought.

If you wanted an LED to come on when the CPU temperature exceeds 50 degrees, the following would currently work:

```python
led = LED(2)
cpu = CPUTemperature(min_temp=50, max_temp=100)

led.source = booleanized(cpu, 0, 1)
```

What if the `booleanized` source tool made assumptions if no `min_value` or `max_value` parameters are provided?

```python
led = LED(2)
cpu = CPUTemperature(min_temp=50, max_temp=100)

led.source = booleanized(cpu)
```

I think there would have to be uniform `min_value` and `max_value` properties perhaps

Alternatively, `booleanized` could instead use `> threshold` instead:

```python
led = LED(2)
cpu = CPUTemperature(threshold=50)

led.source = booleanized(cpu)
```

Obviously this would require `cpu` to be passed in, not some `.values`.

lurch, 2019-04-01:
Yeah, I think your last example is pretty much the same idea I was suggesting above:
```python
led = LED(2)
cpu = CPUTemperature(threshold=50)

led.source = cpu.active_values
```

IMHO this should **definitely** be done with `threshold` though; setting the `min_temp` value to the temperature at which you want it to "turn on" just feels wrong to me - but that's more of a philosophical argument rather than a technical argument :wink: 

> Alternatively, booleanized could instead use `> threshold` instead

Hmmm, perhaps you could change `booleanized` 's logic so that if `min_value` and `max_value` are both `None`, it then reads from an infinite generator of `.is_active` properties (as in your first comment) instead of an infinite generator of `.value` properties? :man_shrugging: