gpiozero/gpiozero #591: Can't run script without env variables.

Kaveshnikov, 2017-08-26:
I was following [this](http://gpiozero.readthedocs.io/en/stable/remote_gpio.html#pin-objects) recipe, but got    this:
```/usr/bin/python3.5 /home/petrarca/remote_gpio/main.py
/usr/local/lib/python3.5/dist-packages/gpiozero/devices.py:451: PinFactoryFallback: Falling back from rpigpio: No module named 'RPi'
  'Falling back from %s: %s' % (name, str(e))))
/usr/local/lib/python3.5/dist-packages/gpiozero/devices.py:451: PinFactoryFallback: Falling back from rpio: No module named 'RPIO'
  'Falling back from %s: %s' % (name, str(e))))
/usr/local/lib/python3.5/dist-packages/gpiozero/devices.py:451: PinFactoryFallback: Falling back from pigpio: failed to connect to localhost:8888
  'Falling back from %s: %s' % (name, str(e))))
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
Can't connect to pigpio at localhost(8888)

Did you start the pigpio daemon? E.g. sudo pigpiod

Did you specify the correct Pi host/port in the environment
variables PIGPIO_ADDR/PIGPIO_PORT?
E.g. export PIGPIO_ADDR=soft, export PIGPIO_PORT=8888

Did you specify the correct Pi host/port in the
pigpio.pi() function? E.g. pigpio.pi('soft', 8888)
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
/usr/local/lib/python3.5/dist-packages/gpiozero/devices.py:451: PinFactoryFallback: Falling back from native: unable to open /dev/gpiomem or /dev/mem; upgrade your kernel or run as root
  'Falling back from %s: %s' % (name, str(e))))
Traceback (most recent call last):
  File "/home/petrarca/remote_gpio/main.py", line 1, in <module>
    from gpiozero import LED
  File "/usr/local/lib/python3.5/dist-packages/gpiozero/__init__.py", line 22, in <module>
    from .devices import (
  File "/usr/local/lib/python3.5/dist-packages/gpiozero/devices.py", line 480, in <module>
    Device.pin_factory = _default_pin_factory()
  File "/usr/local/lib/python3.5/dist-packages/gpiozero/devices.py", line 452, in _default_pin_factory
    raise BadPinFactory('Unable to load any default pin factory!')
gpiozero.exc.BadPinFactory: Unable to load any default pin factory!

Process finished with exit code 1
```
So, is there a way to not use environment variables?

My script:
```from gpiozero import LED
from gpiozero.pins.pigpio import PiGPIOFactory
from time import sleep

remote_factory = PiGPIOFactory(host='192.168.43.20')
led = LED(17, pin_factory=remote_factory)  # remote pin

while True:
    led.off()
    sleep(1)
    led.on()
    sleep(1)
```

bennuttall, 2017-09-04:
This will work without environment variables if you have RPi.GPIO installed, as you do on Raspbian desktop by default, or Raspbian Lite (if you installed the apt package). If, however, you install with pip (either system-wide or in a virtualenv), it won't bring in RPi.GPIO, and no default pin factory will be loaded.

So the documentation is correct, assuming you have RPI.GPIO installed as per Raspbian default.

What happens when you do have RPi.GPIO installed is it happily loads RPi.GPIO as the default pin factory, and then you don't use it but instead manually set your LED's pin factory to a PiGPIO one.

What happens when you don't have RPi.GPIO installed is it looks for RPi.GPIO (and fails), looks for RPIO (and fails), looks for pigpio (and finds it, but as no PIGPIO_ADDR is set, uses 'localhost' and if you don't have pigpiod running locally, it fails).

So to use that script you need one of the following:
- RPi.GPIO installed
- pigpiod running locally
- `PIGPIO_ADDR` environment variable set
- `GPIOZERO_PIN_FACTORY` environment variable set to `mock`

Kaveshnikov, 2017-09-05:
I installed RPI.GPIO but the problem remained.

Herant, 2017-09-09:
When I entering python on my Mac by: 

`$ python`

then I testing if gpiozero works by performing:

`>>> import gpiozero`

the output it gives me is following:

```
/Users/maksimdreggevik/anaconda/envs/myenv3.5/lib/python3.5/site-packages/gpiozero/devices.py:451: PinFactoryFallback: Falling back from rpigpio: No module named 'RPi'
  'Falling back from %s: %s' % (name, str(e))))
/Users/maksimdreggevik/anaconda/envs/myenv3.5/lib/python3.5/site-packages/gpiozero/devices.py:451: PinFactoryFallback: Falling back from rpio: No module named 'RPIO'
  'Falling back from %s: %s' % (name, str(e))))
/Users/maksimdreggevik/anaconda/envs/myenv3.5/lib/python3.5/site-packages/gpiozero/devices.py:451: PinFactoryFallback: Falling back from pigpio: Parent module '' not loaded, cannot perform relative import
  'Falling back from %s: %s' % (name, str(e))))
/Users/maksimdreggevik/anaconda/envs/myenv3.5/lib/python3.5/site-packages/gpiozero/devices.py:451: PinFactoryFallback: Falling back from native: unable to open /dev/gpiomem or /dev/mem; upgrade your kernel or run as root
  'Falling back from %s: %s' % (name, str(e))))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/maksimdreggevik/anaconda/envs/myenv3.5/lib/python3.5/site-packages/gpiozero/__init__.py", line 22, in <module>
    from .devices import (
  File "/Users/maksimdreggevik/anaconda/envs/myenv3.5/lib/python3.5/site-packages/gpiozero/devices.py", line 480, in <module>
    Device.pin_factory = _default_pin_factory()
  File "/Users/maksimdreggevik/anaconda/envs/myenv3.5/lib/python3.5/site-packages/gpiozero/devices.py", line 452, in _default_pin_factory
    raise BadPinFactory('Unable to load any default pin factory!')
gpiozero.exc.BadPinFactory: Unable to load any default pin factory!
```

I followed documentation until 4.3. Environment variables, after that I have no idea what to do.

It tells to :

> The simplest way to use devices with remote pins is to set the PIGPIO_ADDR environment variable to the IP address of the desired Raspberry Pi. You must run your Python script or launch your development environment with the environment variable set using the command line. For example, one of the following:
> 
> $ PIGPIO_ADDR=192.168.1.3 python3 hello.py
> $ PIGPIO_ADDR=192.168.1.3 python3
> $ PIGPIO_ADDR=192.168.1.3 ipython3
> $ PIGPIO_ADDR=192.168.1.3 idle3 &

On my Client PC aka Mac I've got installed Python3.5 under Anaconda env, so I downloaded the whole gpiozero folder from GitHub into site-packages folder there, since the pip installation rather put it in default Mac python environment.

This is getting pretty frustrating, since there is not much help online after I tried to google nearly everything I could think of.

Any idea how to fix this? And if there is please explain how to like you would to a 90 year old grandpa.

Thanks.

bennuttall, 2017-09-15:
@Kaveshnikov What happens when you type `python3 -c "import RPi.GPIO"`?

bennuttall, 2017-09-15:
@Herant Can you make sure pigpiod is running on the remote Pi and try `GPIOZERO_PIN_FACTORY=pigpio PIGPIO_ADDR=192.168.1.3 pinout` ?

Kaveshnikov, 2017-09-15:
@bennuttall I get this traceback:
```python3 -c "import RPi.GPIO"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/local/lib/python3.5/dist-packages/RPi/GPIO/__init__.py", line 23, in <module>
    from RPi._GPIO import *
RuntimeError: This module can only be run on a Raspberry Pi!
```
It's obvious, because I type this on my PC.

bennuttall, 2017-09-18:
Ah, ok. So the issue is just that the documentation assumes you're on a Pi (it only works without environment variables if you're on a Pi). I'll amend that for the next release.

Herant, 2017-09-20:
@bennuttall  if I understood you correctly, I replaced IP address with ip on my rpi, I also changed to "myenv3.5" which is where I git-hub'ed the install files and installed it. And yeah i did sudo pigpiod on rip first.

Now, I ran this code and the answer I am getting is :

(myenv3.5) Users-MacBook-Pro:V0.3 user1$ GPIOZERO_PIN_FACTORY=pigpio PIGPIO_ADDR=192.168.1.130 pinout 0
usage: pinout [-h] [-r REVISION] [-c] [-m]
pinout: error: unrecognized arguments: 0

bennuttall, 2017-09-20:
@Herant You have a `0` on the end of that command. Try without.

Herant, 2017-09-20:
@bennuttall Oh, when u run GPIOZERO_PIN_FACTORY=pigpio PIGPIO_ADDR=192.168.1.130 pinout, I'm getting colored map of gpios. But then I made a new file which I called test.py with following code:

`from gpiozero import LED
from time import sleep

red = LED(17)

while True:
    red.on()
    sleep(1)
    red.off()
    sleep(1)`

which suppose to be test code according to documentation. When I run this code by typing:

(myenv3.5) Users-MBP:Desktop user1$ PIGPIO_ADDR=192.168.1.130 python test.py

I'm getting error:

`/Users/user1/anaconda/envs/myenv3.5/lib/python3.5/site-packages/gpiozero/devices.py:451: PinFactoryFallback: Falling back from rpigpio: No module named 'RPi._GPIO'
  'Falling back from %s: %s' % (name, str(e))))
/Users/user1/anaconda/envs/myenv3.5/lib/python3.5/site-packages/gpiozero/devices.py:451: PinFactoryFallback: Falling back from rpio: No module named 'RPIO.PWM'
  'Falling back from %s: %s' % (name, str(e))))
/Users/user1/anaconda/envs/myenv3.5/lib/python3.5/site-packages/gpiozero/devices.py:451: PinFactoryFallback: Falling back from pigpio: Parent module '' not loaded, cannot perform relative import
  'Falling back from %s: %s' % (name, str(e))))
/Users/user1/anaconda/envs/myenv3.5/lib/python3.5/site-packages/gpiozero/devices.py:451: PinFactoryFallback: Falling back from native: unable to open /dev/gpiomem or /dev/mem; upgrade your kernel or run as root
  'Falling back from %s: %s' % (name, str(e))))
Traceback (most recent call last):
  File "test.py", line 1, in <module>
    from gpiozero import LED
  File "/Users/user1/anaconda/envs/myenv3.5/lib/python3.5/site-packages/gpiozero/__init__.py", line 22, in <module>
    from .devices import (
  File "/Users/user1/anaconda/envs/myenv3.5/lib/python3.5/site-packages/gpiozero/devices.py", line 480, in <module>
    Device.pin_factory = _default_pin_factory()
  File "/Users/user1/anaconda/envs/myenv3.5/lib/python3.5/site-packages/gpiozero/devices.py", line 452, in _default_pin_factory
    raise BadPinFactory('Unable to load any default pin factory!')
gpiozero.exc.BadPinFactory: Unable to load any default pin factory!`

bennuttall, 2017-09-21:
> Oh, when u run GPIOZERO_PIN_FACTORY=pigpio PIGPIO_ADDR=192.168.1.130 pinout, I'm getting colored map of gpios

That means it's working. The pinout information is actually from your remote Pi. Try the same environment variables when running your Python script:

```bash
GPIOZERO_PIN_FACTORY=pigpio PIGPIO_ADDR=192.168.1.130  python3 test.py
```

Herant, 2017-09-22:
@bennuttall when I'm running GPIOZERO_PIN_FACTORY=pigpio PIGPIO_ADDR=192.168.1.130 python3 test.py I getting this:

```
(myenv3.5) Users-MBP:Desktop user1$ GPIOZERO_PIN_FACTORY=pigpio PIGPIO_ADDR=192.168.1.130 python test.py
Traceback (most recent call last):
  File "test.py", line 1, in <module>
    from gpiozero import LED
  File "/Users/user1/anaconda/envs/myenv3.5/lib/python3.5/site-packages/gpiozero/__init__.py", line 22, in <module>
    from .devices import (
  File "/Users/user1/anaconda/envs/myenv3.5/lib/python3.5/site-packages/gpiozero/devices.py", line 480, in <module>
    Device.pin_factory = _default_pin_factory()
  File "/Users/user1/anaconda/envs/myenv3.5/lib/python3.5/site-packages/gpiozero/devices.py", line 455, in _default_pin_factory
    return factory.load()()
  File "/Users/user1/anaconda/envs/myenv3.5/lib/python3.5/site-packages/setuptools-27.2.0-py3.5.egg/pkg_resources/__init__.py", line 2258, in load
  File "/Users/user1/anaconda/envs/myenv3.5/lib/python3.5/site-packages/setuptools-27.2.0-py3.5.egg/pkg_resources/__init__.py", line 2264, in resolve
  File "/Users/user1/anaconda/envs/myenv3.5/lib/python3.5/site-packages/gpiozero/pins/pigpio.py", line 11, in <module>
    import pigpio
  File "/Users/user1/anaconda/envs/myenv3.5/lib/python3.5/site-packages/pigpio.py", line 13, in <module>
    from . import SPI
SystemError: Parent module '' not loaded, cannot perform relative import
```

I tried both with python3 and python, still same output.

waveform80, 2017-09-22:
@Herant how exactly did you install pigpio? From that stack trace, it's claiming that ".../site-packages/pigpio.py" contains "from . import SPI" but the pigpio client library contains no relative imports.

My guess (given that your method of installing gpiozero was to copy the source into your own site-packages), is that you copied `gpiozero/pins/pigpio.py` into site-packages too thinking that was a means of installing pigpio, but I'm afraid it isn't. The [repo for pigpio](https://github.com/joan2937/pigpio/) contains the pigpio.py client library (you can ignore all the C stuff in there - it's mostly for the pigpiod daemon).

However, assuming my guess is correct, I'd suggest just it's a lot easier to just [use conda's built-in pip](https://conda.io/docs/user-guide/tasks/manage-pkgs.html#installing-non-conda-packages) - it should be much simpler than copying stuff manually.

Herant, 2017-09-22:
@waveform80  Well it says that gpiozero is already installed.

```
(myenv3.5) Users-MBP:V0.4 user1$ pip3 install gpiozero pigpio
Requirement already satisfied: gpiozero in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages
Requirement already satisfied: pigpio in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages
(myenv3.5) Users-MBP:V0.4 user1$ pip install gpiozero pigpio
Requirement already satisfied: gpiozero in /Users/user1/anaconda/envs/myenv3.5/lib/python3.5/site-packages
Requirement already satisfied: pigpio in /Users/user1/anaconda/envs/myenv3.5/lib/python3.5/site-packages
(myenv3.5) Users-MBP:V0.4 user1$ 
```

Herant, 2017-09-22:
Alright now it works, all I did was `pip uninstall gpiozero pigpio` and then `pip install gpiozero pigpio`. Thanks for helping @bennuttall  & @waveform80

closed 2018-02-15 by bennuttall

StevenMWirth, 2018-07-20:
Remotely controlling a Pi from Windows 10 using GPIOZero. Ran in to a many of the above problems. Two Windows **User** Environmental Variables cleaned it up. First variable was Variable name: PIGPIO_ADDR, Variable value: 192.168.1.86. Not Variable value: 192.168.1.86 python. My Pi address is 192.168.1.86. Second was Variable name: GPIOZERO_PIN_FACTORY, Variable value: pigpio. This cured the PinFactoryFallback Falling back stuff. Also set the pin numbering scheme to Broadcom (BCM) pin numbering to be consistent with GPIOZero. Beyond that just the install steps from the GPIOZero documentation. Did find not all of the methods in the Recipes are supported. Example no pulse method for LED.