gpiozero/gpiozero #592: 'Edit on GitHub' docs links broken

bennuttall, 2017-09-04:
The 'Edit on GitHub' link in the documentation often tend to be broken.

For example, the link on the [docs homepage](http://gpiozero.readthedocs.io/en/stable/) of the 'stable' branch points to https://github.com/RPi-Distro/python-gpiozero/blob/20e9742f6f0cefa2264914de9dd19ff23312b18e/docs/index.rst which is a broken link.

@waveform80 is there a way we can resolve this? Or can we replace the link with a static link to the github repo?

lurch, 2018-02-21:
Seems like the links work when viewing other versions of the docs, i.e.
* 'v1.4.0' -> https://github.com/RPi-Distro/python-gpiozero/blob/v1.4.0/docs/index.rst
* 'v1.4.1' -> https://github.com/RPi-Distro/python-gpiozero/blob/v1.4.1/docs/index.rst
* 'master' -> https://github.com/RPi-Distro/python-gpiozero/blob/master/docs/index.rst
* 'latest' -> https://github.com/RPi-Distro/python-gpiozero/blob/master/docs/index.rst
* etc.

It's only these that don't work:
* 'stable' -> https://github.com/RPi-Distro/python-gpiozero/blob/6b9ca10eee7e27313b94aa46651a9fcc412bbc95/docs/index.rst
* 'pinout-man' -> https://github.com/RPi-Distro/python-gpiozero/blob/pinout-man/docs/index.rst

So maybe RTD needs to be told to ignore particular 'versions' / tags / branches ?
Is the 'stable' docs version supposed to effectively be a 'symlink' of the latest released version, i.e. 'v1.4.1' at the time of writing?

SteveAmor, 2018-03-27:
@bennuttall if you login to http://readthedocs.org/projects/gpiozero/ are you able to set the ```Default Version``` to ```latest```?

bennuttall, 2018-03-27:
I am - but I'd prefer the main version people see is the latest that's been released, rather than the latest on github. It could be that we implement and document new features (or make changes to existing classes), which people read about in the docs, but haven't been released yet.

SteveAmor, 2018-03-28:
Ah, yes, of course.
It's very odd because the RTD link to the current "stable" commit SHA-1 ```824caefc0a780a68723596275ea3019796d704dc``` (and previous SHA-1's above) doesn't exist at all on any GitHub repo.
I guess it must be a commit on a branch on a local git repo which a ```git show 824caefc0a780a68723596275ea3019796d704dc``` would highlight.

SteveAmor, 2018-03-28:
This is a known problem over at RTD https://github.com/rtfd/readthedocs.org/issues/1820.
The SHA-1 is for a tag, not the commit.
On my local repo:
``` codeblock
~/python-gpiozero $ git show 824caefc0a780a68723596275ea3019796d704dc
tag v1.4.1
Tagger: Dave Jones <dave@waveform.org.uk>
Date:   Wed Feb 21 15:35:25 2018 +0000

Release v1.4.1
```

SteveAmor, 2018-03-28:
The RTD PR to fix this is still waiting to be merged on the production servers.
https://github.com/rtfd/readthedocs.org/pull/3615

bennuttall, 2018-03-28:
:+1: 

I'll close this as it's out of out control, and in hand.

closed 2018-03-28 by bennuttall

lurch, 2018-03-28:
@bennuttall see https://github.com/rtfd/readthedocs.org/issues/1820#issuecomment-375833634 (which suggests rebuilding the docs _should_ now fix it)

bennuttall, 2018-03-28:
Hmm I've rebuilt stable but it doesn't seem to have changed the link

SteveAmor, 2018-03-28:
I believe RTD have still got to merge https://github.com/rtfd/readthedocs.org/pull/3615 which tells their production workflow to use git 2.  There servers are now running git 2.7.4 (as of 5 days ago) but they still need to update their workflow.