gpiozero/gpiozero #593: Button with external pullup/pulldown resistor

jdimpson, 2017-09-07:
**Feature Request**:

I have need of a `gpiozero.Button`  or `gpiozero.DigitalInputDevice` object that doesn't pull up or pull down the pin via the internal resistors in the BCM chip, so that instead I can use external resistors to do so.

In my case, I have a voltage divider on the RPi pin to protect it from overvoltage. I am also using very high resistance to slow down the voltage divider's drain on my battery. I need to detect when the voltage on the pin goes to ground.

I've successfully utilized the following `RPi.GPIO` code for my application. Note lack of `pull_up_down` attribute in `GPIO.setup()`.
```
import RPi.GPIO as GPIO
def poweroff():
        from subprocess import call
        print("shutting down")
        call(["sudo","poweroff"])
slee=10
pin=19
GPIO.setmode(GPIO.BCM)
GPIO.setup(pin, GPIO.IN)
while True:
        val = bool(GPIO.input(pin))
        if not val:
                print "pin {p} reads {v}".format(p=pin,v=val)
                poweroff()
        sleep(slee)
```

If instead I call `GPIO.setup(pin, GPIO.IN,pull_up_down=GPIO.PUD_UP)`, then `val` always evaluates `True` regardless of the voltage on the pin, and if I call `GPIO.setup(pin, GPIO.IN,pull_up_down=GPIO.PUD_DOWN)`, `val` always evaluates `False`. This is because (according to my current theory) since the internal resistor is at 100 kiloohms, it offers much less resistance than the R2 (to ground) resistor on my voltage divider. So the pin never sees the voltage go to zero because the internal resistor remains the lowest resistance to ground or high.

My reading of the documentation for `gpiozero.Button` is that `pull_up` is asserted (`True`) by default and the pin is pulled up; if `pull_up` is set to `False`, then the pin is pulled down.  So I expect it to work like it does in the above `RPi.GPIO` example when pulled up or down, which won't work for my application. The documentation for `gpiozero.DigitalInputDevice` shows `pull_up` is a valid argument in the constructor, but doesn't discuss it further; I assume it has the same behaviour as Button. So it seems in the API, there's no way to avoid using an internal up/down resistor.

Is this reading correct? If so, is it reasonable to ask for a way for the Button or DigitialInputDevice class to neither pull up or pull down, to accommodate for use of external pull up/down resistors? (Maybe by setting `pull_up=None`?)

I can certainly continue to use RPi.GPIO, but I like gpiozero and want to use it more.

**Background**: 

I am trying to read a low battery (LB) notification signal from a LiPoly voltage booster without having to also add an ADC or active voltage reduction circuitry to the RPi. The LB signal is at battery voltage until battery dips to 3.2 Volts, at which point the LB signal goes to nearly zero (a few millivolts). When the battery is full, the LB is at 4.18 Volts, thus the need for the divider.

My Voltage Divider resistor R1, from the LB signal source to RPi pin, is 1.5 megaohms; resistor R2, from RPi pin to ground, is 5.6 megaohms. The LB signal may be anywhere between 3.2 to 4.18 volts, but will drop to nearly 0 when indicating Low Battery state. 

Alternatively, from experimentation it seems I could avoid the Voltage Divider and just put a very high resistor from the LB signal to the RPi pin to drop the voltage. This avoids bringing the pin to ground and the associated power drain (no matter how small it is). A 5.6 Megaohm resister seems to cause about 1 volt of drop, which seems sufficient. But 1) I'm not sure if doing that is a safe/reliable way to ensure reduced voltage, and 2) I realize this isn't the forum for general electrical engineering questions. :)

bennuttall, 2017-09-15:
It's perfectly valid to want to use no pull-up or pull-down. We probably should add `pull_up=None` for this case.

lurch, 2017-11-30:
@jdimpson In the meantime, you might be able to work around your problem by using [GPIODevice](https://gpiozero.readthedocs.io/en/stable/api_input.html#gpiodevice) instead of DigitalInputDevice.
Or looking at https://gpiozero.readthedocs.io/en/stable/api_pins.html#gpiozero.Pin.pull you _might_ be able to do `button.pin.pull = 'floating'` after creating your `button` object?

@bennuttall [InputDevice](https://gpiozero.readthedocs.io/en/stable/_modules/gpiozero/input_devices.html#InputDevice) (which is where everything inherits the `pull_up` attribute from) also uses the value of `pull_up` to determine whether the input is active-high or active-low, so if we allow `pull_up=None` then I guess we'd need some other means of specifying this :-/

Naohiro2g, 2019-01-30:
Hi Ben, do you have any conflicts around this? We eagerly want to use the external resistors in any value for wider applications.

I understand it's kind of a big change to have use-internal_pulls instead of pull_up=None, even if we want to be more logical.

bennuttall, 2019-01-30:
I don't have any issue - it should be addressed - I just want @waveform80 to review PR #658.

PR #658 (2019-02-02): Allow input devices to be pulled up or down externally, close #593
Fix #593
commit 0639b4c1 (2019-02-02): Allow input devices to be pulled up or down externally

Fix #593
closed 2019-02-02 by waveform80