gpiozero/gpiozero #598: Add Maplin Remote Controlled Mains Switches - help getting started writing the class

SteveAmor, 2017-10-29:
@bennuttall @waveform80 @lurch I hope you can help me :-)

I want to write a class to control the following remote control sockets from Maplin:

https://www.maplin.co.uk/p/remote-controlled-mains-socket-set-single-n78ka

They need a 433MHz transmitter to operate them.  You can buy these cheaply from eBay:

https://www.ebay.co.uk/itm/433MHz-RF-Transmitter-and-Receiver-Pair-for-Pi-Arduino-ASK-OOK-Remote-Wireless/181863740559

This is somewhat similar to the Energenie socket but you can put the 433MHz transmitter on any GPIO pin you choose and you could have multiple 433MHz transmitters  (to make it more generic, let’s call them OOKTX as the radio frequency is a function of the transmitter, not gpiozero).

I have fully working code but it’s not OOP and I’d really appreciate some help to get me started to add the code to gpiozero.  Firstly, which existing classes to extend and secondly (but less important) the class names.  I’m thinking of the following:

OOKTX extends Device and represents an On Off Key (OOK) transmitter.

class gpiozero.OOKTX(pin, *, active_high=True, initial_value=False, period=0.00001, pin_factory=None)

RadioSocket extends ??Device??  and represents a remote controlled mains socket. 

class gpiozero.RadioSocket(OOKTX=None, *, socket=None, active_high=True, 
initial_value=False)

Methods for the RadioSocket would be the same as the Energenie sockets but in this case, there are 16 sockets.

I’m really struggling with the linkage between the class OOKTX and class RadioSocket.

bennuttall, 2017-11-03:
Sounds great!

SteveAmor, 2017-11-04:
I will probably have the following (publicly accessible) classes:
OOK (extends SharedMixin, OutputDevice)
RemoteSocket (extends Device) - so that other types of remote sockets can have classes
MaplinRemoteSocket (extends RemoteSocket) - one type of many 433Mhz OOK radio controlled sockets

bennuttall, 2018-09-25:
Are these sold anywhere else now Maplin is gone?

SteveAmor, 2018-10-17:
These look to be the same https://www.amazon.co.uk/gp/product/B0053XRZ9I

SteveAmor, 2018-10-26:
Let's see if Maplin online sell them :-)