gpiozero/gpiozero #600: "import gpiozero" shouldn't fail without a valid pin factory

bennuttall, 2017-10-30:
The following example is a valid method of setting the pin factory to mock:

```python
import gpiozero
from gpiozero.pins.mock import MockFactory
gpiozero.Device.pin_factory = MockFactory()
```

but it only works if `import gpiozero` is successful, having found a valid pin factory.

If you're on a Raspberry Pi, that's easy - you just install rpi.gpio (even if you don't intend to use it).

If you're on a PC, the following will happen:

- rpi.gpio will fail (even if it's installed - can only be run on a Pi)
- rpio will fail (even it it's installed - can only be run on a Pi)
- pigpio will fail (even if it's installed - pigpiod not running on localhost)
- native will fail (can't access /dev/mem)

And it runs out of options.

The only way to get it to work is to set the pin factory in an environment variable.

It would be nice if you could get past `import gpiozero`, and if no pin factory is found, it's set to `None` and fails when you try to create a device. Would that be that possible?

bennuttall, 2017-10-30:
This would also resolve the issue with `pinout` on PCs needing pin factory set.

drewcassidy, 2018-02-20:
This should probably be explained in the documentation in the meantime, I spent a while trying to figure out why the example code [here](http://gpiozero.readthedocs.io/en/stable/api_pins.html#mock-pins) was throwing errors when trying to run on my laptop without a pi connected.

devxpy, 2018-03-30:
@drewcassidy. I am frustated with this thing. Why does a mock exist, when you can't even use it on a PC?

```
In [3]: from gpiozero.pins.mock import MockFactory
/home/dev/.pyenv/versions/3.6.4/lib/python3.6/site-packages/gpiozero/devices.py:452: PinFactoryFallback: Falling back from rpigpio: No module named 'RPi'
  'Falling back from %s: %s' % (name, str(e))))
/home/dev/.pyenv/versions/3.6.4/lib/python3.6/site-packages/gpiozero/devices.py:452: PinFactoryFallback: Falling back from rpio: No module named 'RPIO'
  'Falling back from %s: %s' % (name, str(e))))
/home/dev/.pyenv/versions/3.6.4/lib/python3.6/site-packages/gpiozero/devices.py:452: PinFactoryFallback: Falling back from pigpio: No module named 'pigpio'
  'Falling back from %s: %s' % (name, str(e))))
/home/dev/.pyenv/versions/3.6.4/lib/python3.6/site-packages/gpiozero/devices.py:452: PinFactoryFallback: Falling back from native: unable to open /dev/gpiomem or /dev/mem; upgrade your kernel or run as root
  'Falling back from %s: %s' % (name, str(e))))
---------------------------------------------------------------------------
BadPinFactory                             Traceback (most recent call last)
<ipython-input-3-456f994e72a8> in <module>()
----> 1 from gpiozero.pins.mock import MockFactory

~/.pyenv/versions/3.6.4/lib/python3.6/site-packages/gpiozero/pins/mock.py in <module>()
     26     PinInvalidPull,
     27     )
---> 28 from ..devices import Device
     29 from .pi import PiPin
     30 from .local import LocalPiFactory

~/.pyenv/versions/3.6.4/lib/python3.6/site-packages/gpiozero/__init__.py in <module>()
     20 # contamination here ... and besides, have you *seen* the list lately?!
     21 from .exc import *
---> 22 from .devices import (
     23     Device,
     24     GPIODevice,

~/.pyenv/versions/3.6.4/lib/python3.6/site-packages/gpiozero/devices.py in <module>()
    484 
    485 
--> 486 Device.pin_factory = _default_pin_factory()
    487 atexit.register(_shutdown)

~/.pyenv/versions/3.6.4/lib/python3.6/site-packages/gpiozero/devices.py in _default_pin_factory(name)
    451                     PinFactoryFallback(
    452                         'Falling back from %s: %s' % (name, str(e))))
--> 453         raise BadPinFactory('Unable to load any default pin factory!')
    454     else:
    455         # Try with the name verbatim first. If that fails, attempt with the

BadPinFactory: Unable to load any default pin factory!

```

bennuttall, 2018-03-30:
> I am frustated with this thing. Why does a mock exist, when you can't even use it on a PC?

@devxpy It does. You just have to set it in an environment variable, like I said above.

I see you're running on Linux so you should be able to launch IPython with the env var set in-line:

```bash
$ GPIOZERO_PIN_FACTORY=mock ipython
```

or export it:

```bash
$ export GPIOZERO_PIN_FACTORY=mock
$ ipython
```

or add it to your `.bashrc` or equivalent.

This issue points out that it's not currently possible to set the pin factory to mock if the default pin factory fails to load. There are situations where it does, like on a Pi, but as you have also discovered, there are situations where it doesn't, and it's impossible to import without a fatal error.

For now, this is easily resolved by setting the pin factory to mock in an environment variable, and we'll try to fix it for the next release.

devxpy, 2018-03-31:
@bennuttall. I'm sorry, i was in a virtualenv and was exporting outside the virtualenv

bennuttall, 2018-10-02:
How about if importing gpiozero without a valid pin factory just shows a warning?

```python
>>> import gpiozero
Warning: No default pin factory found. Please set gpiozero.Device.pin_factory
>>> led = LED(2)
InvalidPinFactory: No default pin factory set
>>> gpiozero.Device.pin_factory = MockFactory()
>>> led = LED(2)
>>> led
<gpiozero.LED object on pin GPIO2, active_high=True, is_active=False>
```

bennuttall, 2018-10-04:
@waveform80 agrees - let's get this in v1.5

clach04, 2018-10-10:
In case anyone else hits this, hacky workaround:

    import os
    os.environ['GPIOZERO_PIN_FACTORY'] = os.environ.get('GPIOZERO_PIN_FACTORY', 'mock')
    import gpiozero

NOTE `from gpiozero.pins.mock import MockFactory` as documented in https://gpiozero.readthedocs.io/en/stable/api_pins.html#mock-pins isn't needed with this (nor does it work without this). Setting the explicit mock factory is not required.

bennuttall, 2019-01-25:
Note that if/when this gets fixed, the note I added to the docs will need removing: https://gpiozero.readthedocs.io/en/latest/remote_gpio.html#pin-factories

closed 2019-01-27 by bennuttall

lurch, 2019-02-10:
> Note that if/when this gets fixed, the note I added to the docs will need removing: https://gpiozero.readthedocs.io/en/latest/remote_gpio.html#pin-factories

Seems that #722 forgot to do that? :wink:

PR #722 (2019-01-27): I said import, damn it!
Permit gpiozero to be imported without trying to load a pin factory; only load a pin factory on first Device construction. While we're at it re-work the default pin factory loader to avoid pkg_resources import in as many cases as possible (while keeping it around for third-party pin implementations). This is just because pkg_resources takes *ages* to load (especially on an I/O limited platform like the Pi).

Fixes #600 and #675
commit 8b3671cb (2019-01-27): I said import, damn it! (#722)

* Fix #600, fix #675

Permit gpiozero to be imported without trying to load a pin factory;
only load a pin factory on first Device construction. While we're at it
re-work the default pin factory loader to avoid pkg_resources import in
as many cases as possible (while keeping it around for third-party pin
implementations). This is just because pkg_resources takes *ages* to
load (especially on an I/O limited platform like the Pi).

* Add get-pip.py to gitignore

Tox creates it for the 3.2 and 3.3 envs

* Support ancient pytest versions

The yield form of pytest fi [... cut]