gpiozero/gpiozero #601: Add support to use BOARD mode

filips123, 2017-11-03:
Add support to use BOARD mode. BCM mode should be default. Mode could be at the beginning of the program (as in RPi.GPIO) or for each device separately (`led = LED(7, mode=BOARD)`).

bennuttall, 2017-11-03:
Duplicate of #467.

This has been debated back-and-forth. We distinctly only supported a single numbering system in the beginning, to keep things simple. Your best bet is to use a port label or the [pinout command line tool](http://gpiozero.readthedocs.io/en/stable/cli_tools.html#pinout).

closed 2017-11-03 by bennuttall

filips123, 2017-11-03:
If you were to support both ways, but one would be default, it would not be more complicated.
In addition, I am interested in how to support the older versions of Raspberry Pi, which have a different labeling without changes.

bennuttall, 2017-11-03:
But any resources people make will vary - some using one numbering system, some using another.

GPIO Zero works with any Pi model, as BCM pin numbers work wherever they're located on the board. And the `pinout` tool will show you an accurate pinout diagram for the Pi you're on.