gpiozero/gpiozero #605: Error when used with MCP3008

dpritt, 2017-11-19:
Ben,

I have a couple of PIs with MCP3008 analogue devices, both are throwing the following exception.
Python 3.5 running on Stretch.

Exception ignored in: <bound method GPIOBase.__del__ of <gpiozero.MCP3008 object using <weakproxy at 0xb3b35d20 to NoneType at 0x3e5788>>>
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 122, in __del__
    self.close()
  File "/usr/lib/python3/dist-packages/gpiozero/spi_devices.py", line 37, in close
    if self._spi:
ReferenceError: weakly-referenced object no longer exists

Any suggestions on how to stop this?

Regards

Dave

bennuttall, 2017-11-23:
Can you post the code you're using? (please use fenced code blocks)

I have seen other people report of SPI not working on Stretch but I can't reproduce. The following works fine for me:

```python
from gpiozero import MCP3008

pot = MCP3008()

while True:
    print(pot.value)
```

dpritt, 2017-11-26:
Ben,

This is part of my Solar heating and pool manager system, I use the MCP3008 to read the thermistors on the solar array and the water temperature as it passes through the filter.

The code for the MCP3008 part is very simple

I load the libraries

```

from gpiozero import MCP3008

# Constants 

vref = 3.3

# Variables used for the ADC channels

pool_channel = 0                        # A0
solar_channel = 1                       # A1

```
The code that reads the ADC, calls the sub-modules to convert it and store the values in an array is below, this is run every 30 seconds as part of the main loop that controls the pump speeds and diverter valves, reports values back etc.

```
            # Read the Pool temperature sensor data 
            adc = MCP3008(channel=pool_channel)
            pool_volts = vref * adc.value
            pool_res   = ConvertResistance(pool_volts,0)
            pool_temp  = ConvertTemp(pool_volts)

            # Enter the reading in the next position in the array 
            np.put(a_p,a_index,pool_temp)                        
         
            # Read the Array temperature sensor data 
            adc = MCP3008(channel=solar_channel)
            solar_volts = vref * adc.value
            solar_res   = ConvertResistance(solar_volts,0)
            solar_temp  = ConvertTemp(solar_volts)

            #  Enter the reading in the next position in the array 
            np.put(a_s,a_index,solar_temp)

```
                                      
Nothing special at all about what I am doing here.

The error messages come from the gpiozero code.

Regards

Dave

Can you post the code you're using? (please use fenced code blocks)
I have seen other people report of SPI not working on Stretch but I can't reproduce. The following works fine for me:
from gpiozero import MCP3008

pot = MCP3008()

while True:
    print(pot.value)
—
This email was Anti Virus checked by Astaro Security Gateway. http://www.sophos.com

gkovacs81, 2018-01-10:
I have a similar problem in my project. When the application stops I get three exceptions:
Exception ignored in: <bound method MCP3008.__del__ of <gpiozero.MCP3008 object using <weakproxy at 0xb55e5900 to LocalPiSoftwareSPIShared at 0xb5541770>>>
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 122, in __del__
    self.close()
  File "/usr/lib/python3/dist-packages/gpiozero/spi_devices.py", line 38, in close
    self._spi.close()
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 82, in close
    old_close()
  File "/usr/lib/python3/dist-packages/gpiozero/pins/local.py", line 175, in close
    super(LocalPiSoftwareSPI, self).close()
  File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 77, in close
    super(SourceMixin, self).close()
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 393, in close
    self._pin.close()
  File "/usr/lib/python3/dist-packages/gpiozero/pins/rpigpio.py", line 106, in close
    GPIO.cleanup(self.number)
RuntimeError: Please set pin numbering mode using GPIO.setmode(GPIO.BOARD) or GPIO.setmode(GPIO.BCM)
Exception ignored in: <bound method MCP3008.__del__ of <gpiozero.MCP3008 object using SPI(closed)>>
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 122, in __del__
    self.close()
  File "/usr/lib/python3/dist-packages/gpiozero/spi_devices.py", line 38, in close
    self._spi.close()
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 82, in close
    old_close()
  File "/usr/lib/python3/dist-packages/gpiozero/pins/local.py", line 175, in close
    super(LocalPiSoftwareSPI, self).close()
  File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 77, in close
    super(SourceMixin, self).close()
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 393, in close
    self._pin.close()
  File "/usr/lib/python3/dist-packages/gpiozero/pins/rpigpio.py", line 106, in close
    GPIO.cleanup(self.number)
RuntimeError: Please set pin numbering mode using GPIO.setmode(GPIO.BOARD) or GPIO.setmode(GPIO.BCM)
Exception ignored in: <bound method LocalPiSoftwareSPIShared.__del__ of SPI(closed)>
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 137, in __del__
    super(SharedMixin, self).__del__()
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 122, in __del__
    self.close()
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 82, in close
    old_close()
  File "/usr/lib/python3/dist-packages/gpiozero/pins/local.py", line 175, in close
    super(LocalPiSoftwareSPI, self).close()
  File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 77, in close
    super(SourceMixin, self).close()
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 393, in close
    self._pin.close()
  File "/usr/lib/python3/dist-packages/gpiozero/pins/rpigpio.py", line 106, in close
    GPIO.cleanup(self.number)
RuntimeError: Please set pin numbering mode using GPIO.setmode(GPIO.BOARD) or GPIO.setmode(GPIO.BCM)

I wrote a basic application to reproduce the problem:

mcp_adapter.py:
from gpiozero import MCP3008
class Adapter(object):

    def __init__(self):
        self.pot = MCP3008()
        self.p2 = MCP3008()
        self.p3 = MCP3008()

    def get_value(self):
        return self.pot.value

monitor.py:
from time import sleep
from threading import Thread
from mcp_adapter import Adapter

class Monitor(Thread):

    def __init__(self, stop):
        super(Monitor, self).__init__()
        self.a = Adapter()
        self.st = stop

    def run(self):
        while self.st.is_set():
            print(self.a.get_value())
            sleep(1)

worker.py:
from monitor import Monitor
from threading import Event

stop = Event()
m = Monitor(stop)
m.start()

while True:
    try:
        print("test-")
    except KeyboardInterrupt:
        print("exit")
m.join()

sys.exit(1)

If it is less complex the exception doesn't appear.

Environment:
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 8.0 (jessie)
Release:        8.0
Codename:       jessie

Package: python-gpiozero
Source: gpiozero
Version: 1.4.0

gkovacs81, 2018-09-22:
from gpiozero import MCP3008

m = MCP3008(channel=3, clock_pin=21, mosi_pin=16, miso_pin=20, select_pin=1)
m = MCP3008(channel=2, clock_pin=21, mosi_pin=16, miso_pin=20, select_pin=1)

===>
0.0004885197850512668
0.0004885197850512668
Exception ignored in: <bound method GPIOBase.__del__ of <gpiozero.MCP3008 object using <weakproxy at 0xb6180660 to NoneType at 0x3e5788>>>
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 122, in __del__
  File "/usr/lib/python3/dist-packages/gpiozero/spi_devices.py", line 37, in close
ReferenceError: weakly-referenced object no longer exists

bennuttall, 2018-09-22:
What's happening is you're creating a reference to an SPI object, and then throwing it away by creating another with the same name. They're supposed to be connected, as they use a shared bus. But if you throw one away you have problems.

You need to use separate variable names (e.g. `m1`, `m2`):

```python
m1 = MCP3008(channel=3, clock_pin=21, mosi_pin=16, miso_pin=20, select_pin=1)
m2 = MCP3008(channel=2, clock_pin=21, mosi_pin=16, miso_pin=20, select_pin=1)

while True:
    print(m1.value, m2.value)
```

gkovacs81, 2018-09-23:
OK, another mistake not using the variable 'i'!

```python
from gpiozero import MCP3008

sensors = []
for i in range(0,6):
    sensors.append(MCP3008(channel=i, clock_pin=21, mosi_pin=16, miso_pin=20, select_pin=1))

for s in sensors:
    print(s.value)
```

0.0004885197850512668
0.002442598925256556
0.0014655593551538004
0.5007327796775769
0.002442598925256556
0.0014655593551538004
Exception ignored in: <bound method GPIOBase.__del__ of <gpiozero.MCP3008 object using SPI(closed)>>
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 122, in __del__
    self.close()
  File "/usr/lib/python3/dist-packages/gpiozero/spi_devices.py", line 38, in close
    self._spi.close()
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 82, in close
    old_close()
  File "/usr/lib/python3/dist-packages/gpiozero/pins/local.py", line 173, in close
    super(LocalPiSoftwareSPI, self).close()
  File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 77, in close
    super(SourceMixin, self).close()
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 394, in close
    self._pin.close()
  File "/usr/lib/python3/dist-packages/gpiozero/pins/rpigpio.py", line 106, in close
    GPIO.cleanup(self.number)
RuntimeError: Please set pin numbering mode using GPIO.setmode(GPIO.BOARD) or GPIO.setmode(GPIO.BCM)
Exception ignored in: <bound method GPIOBase.__del__ of <gpiozero.MCP3008 object using <weakproxy at 0xb6141360 to LocalPiSoftwareSPIShared at 0xb6022810>>>
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 122, in __del__
    self.close()
  File "/usr/lib/python3/dist-packages/gpiozero/spi_devices.py", line 38, in close
    self._spi.close()
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 82, in close
    old_close()
  File "/usr/lib/python3/dist-packages/gpiozero/pins/local.py", line 173, in close
    super(LocalPiSoftwareSPI, self).close()
  File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 77, in close
    super(SourceMixin, self).close()
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 394, in close
    self._pin.close()
  File "/usr/lib/python3/dist-packages/gpiozero/pins/rpigpio.py", line 106, in close
    GPIO.cleanup(self.number)
RuntimeError: Please set pin numbering mode using GPIO.setmode(GPIO.BOARD) or GPIO.setmode(GPIO.BCM)
Exception ignored in: <bound method SharedMixin.__del__ of SPI(closed)>
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 137, in __del__
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 122, in __del__
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 82, in close
  File "/usr/lib/python3/dist-packages/gpiozero/pins/local.py", line 173, in close
  File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 77, in close
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 394, in close
  File "/usr/lib/python3/dist-packages/gpiozero/pins/rpigpio.py", line 106, in close
RuntimeError: Please set pin numbering mode using GPIO.setmode(GPIO.BOARD) or GPIO.setmode(GPIO.BCM)

retroriff, 2019-01-29:
I have the same issue when I stop a script which reads multiple MCP3008 devices. It only gives the errror when using non SPI pins, for example:

`for i in range(4): 
device1[i] = MCP3008(channel=i, clock_pin=21, mosi_pin=20, miso_pin=19, select_pin=7)`

It works using default SPI pins (gpio8, gpio9, gpio10, gpio11)

`device1[i] = MCP3008(channel=i, clock_pin=21, mosi_pin=20, miso_pin=19, select_pin=7)`

lurch, 2019-02-11:
Re-opening and assigning to @waveform80 , as he wrote all the SoftwareSPI stuff :slightly_smiling_face:

waveform80, 2019-02-21:
Yup, looks like there's definitely something iffy in the life-span of stuff using `SPISoftwareBus` (which doesn't surprise me too much - I vaguely recall there's some very quick hacks around the pin reservation stuff in there - probably time it got sorted out "properly"). Useful to know that it's not occurring with the hardware-driven (spidev) implementation though.

I'll dig into this...