gpiozero/gpiozero #607: Nested ButtonBoard?

bennuttall, 2017-11-29:
It's possible to nest LEDBoards within LEDBoards, but the same does not apply to ButtonBoard. If there's no reason this can't exist, we should add it, and include it in tests in #606.

lurch, 2017-12-01:
Should be possible, I guess it'd just need to have all the `isinstance` related code that http://gpiozero.readthedocs.io/en/stable/_modules/gpiozero/boards.html#LEDCollection uses (except in this case it'd be using `isinstance(pin_or_collection, ButtonBoard)` instead of `isinstance(pin_or_collection, LEDCollection)` and `Button` instead of `LEDClass`).

Coincidentally I ordered myself a ScoreZero earlier today, so I'll see if I get time over the weekend to have a fiddle...

lurch, 2017-12-01:
Hmmm, although it might need a bit of thinking about how events are hooked up, and whether changes in a sub-button-board also bubble up to the parent button-board...