gpiozero/gpiozero #608: Another PiTraffic board

lurch, 2017-11-30:
While browsing through Amazon, I just spotted another interesting-looking board calling itself PiTraffic https://www.amazon.co.uk/Educational-SB-Components-crowdfunded-Kickstarter/dp/B073NRL497

Unfortunately GpioZero already contains a [PiTraffic class](https://gpiozero.readthedocs.io/en/stable/api_boards.html#pi-traffic) for another board, and I have no good ideas right now about how to resolve this naming conflict.

I found some example code for it at https://github.com/sbcshop/PiTraffic and based on that (and using https://pinout.xyz/ to convert BOARD numbers to BCM numbers), adding support for it to GpioZero _should_ be as simple as:
```python
class NewPiTraffic(CompositeOutputDevice):
    def __init__(self, pwm=False, pin_factory=None):
        super(NewPiTraffic, self).__init__(
            north=TrafficLights(5, 6, 13, pwm=pwm, pin_factory=pin_factory),
            east=TrafficLights(16, 20, 21, pwm=pwm, pin_factory=pin_factory),
            south=TrafficLights(17, 27, 22, pwm=pwm, pin_factory=pin_factory),
            west=TrafficLights(23, 24, 25, pwm=pwm, pin_factory=pin_factory),
            buzzer=Buzzer(18, pin_factory=pin_factory),
            button=Button(4, pull_up=True, pin_factory=pin_factory),
            _order=('north', 'east', 'south', 'west', 'buzzer', 'button'),
            pin_factory=pin_factory
        )
```
which would then allow you to do e.g.
```python
trafficboard = NewPiTraffic()
trafficboard.east.green.on()
trafficboard.north.amber.blink()
sleep(5)
trafficboard.east.off()
trafficboard.north.off()
```

Alternatively, we could create a sub-device grouping all the traffic-lights together like this:
```python
class NewPiTraffic(CompositeOutputDevice):
    def __init__(self, pwm=False, pin_factory=None):
        super(NewPiTraffic, self).__init__(
            lights=CompositeOutputDevice(
                north=TrafficLights(5, 6, 13, pwm=pwm, pin_factory=pin_factory),
                east=TrafficLights(16, 20, 21, pwm=pwm, pin_factory=pin_factory),
                south=TrafficLights(17, 27, 22, pwm=pwm, pin_factory=pin_factory),
                west=TrafficLights(23, 24, 25, pwm=pwm, pin_factory=pin_factory),
                _order=('north', 'east', 'south', 'west'),
                pin_factory=pin_factory
            ),
            buzzer=Buzzer(18, pin_factory=pin_factory),
            button=Button(4, pull_up=True, pin_factory=pin_factory),
            _order=('lights', 'buzzer', 'button'),
            pin_factory=pin_factory
        )
```
which would then allow you to do e.g.
```python
trafficboard = NewPiTraffic()
trafficboard.lights.east.green.on()
trafficboard.lights.north.amber.blink()
sleep(5)
trafficboard.lights.off()
```

I don't have a PiTraffic, so none of these code-examples have been tested.

Ping @sbcshop

sbcshop, 2017-11-30:
> Educational-SB-Components-crowdfunded-Kickstarter/dp/B073NRL497

Hi,

I have forwarded this to our IT developer.

Will come to your answer.

GAJENDER SINGH
DIRECTOR
<02035140914>02035140914
<#UNIQUE_ID_SafeHtmlFilter_UNIQUE_ID_SafeHtmlFilter_UNIQUE_ID_SafeHtmlFilter_UNIQUE_ID_SafeHtmlFilter_>
gajender.singh@sb-components.co.uk
https://shop.sb-components.co.uk
<https://twitter.com/sb_ltd> <https://www.facebook.com/sbcomponentsltd>
<https://www.youtube.com/channel/UC-HPa0xV9LtjqT79aajhGfQ>
<https://plus.google.com/115397217718246801767>
<https://shop.sb-components.co.uk>
Unit B, 58B Kingsbridge Crescent,
Southall,  UB1 2DL  United Kingdom
Vat: GB154489384

This electronic transmission is strictly confidential and intended solely
for the addressee. It may contain information which is covered by legal,
professional or other privilege. If you are not the intended addressee, you
must not disclose, copy or take any action in reliance of this
transmission. If you have received this transmission in error, please
delete it and notify us as soon as possible. The Internet cannot guarantee
the integrity of this message, SB Components Ltd will not therefore be
liable for the message if modified.

bennuttall, 2017-11-30:
I was thinking about adding this when I saw it recently.

I actually assumed it was a 4Tronix product, but it looks like they make a different (albeit similar) board altogether: https://shop.4tronix.co.uk/products/pi-stop-road-crossing-base-for-pistops-on-raspberry-pi

It would be nice to add support for both of these.

It's annoying that there's a name conflict. I'm not sure if the SB one is a genuine HAT or not, but we could call it `PiTrafficHat` (although we already have a `TrafficHat` - we might have to call it `SBPiTraffic` or something (similarly, `PiHutXmasTree` was suggested) in #502.

The 4Tronix one is just called "Pi-Stop Crossing" so could be `PiStopCrossing`.

While they are obviously very similar in principle, they do differ in on-board components: SB has a button and a buzzer, whereas the 4Tronix one has 4 buttons and a buzzer. I can't see any example code (and not sure what names are generally given to the four quadrants), so not sure on naming convention there. But I'm sure there can be commonality, as there is with the poorly named internal-ish `TrafficLightsBuzzer` class :)

lurch, 2017-11-30:
> I'm not sure if the SB one is a genuine HAT or not

For such a simple and low-cost board, I'd be **very** surprised if it contains the HAT-mandated EEPROM chip.

> While they are obviously very similar in principle, they do differ in on-board components: SB has a button and a buzzer, whereas the 4Tronix one has 4 buttons and a buzzer.

The differences appear to go deeper than just the number of buttons - looking at the pinout-info at the bottom of the 4tronix link you posted, it seems that the PiStopCrossing actually 'doubles up' the up/down and left/right signals for both the traffic lights, the pedestrian lights and the buttons. So in code, it _might_ look something like:
```python
class _PiStopCrossingSegment(CompositeOutputDevice):
    def __init__(self, trafficlights, pedestrianlights, button, pin_factory=None):
        super(_PiStopCrossingSegment, self).__init__(
            trafficlights=trafficlights,
            pedestrianlights=pedestrianlights,
            button=button,
            order=(trafficlights, pedestrianlights, button),
            pin_factory=pin_factory
        )

class PiStopCrossing(CompositeOutputDevice):
    def __init__(self, pwm=False, pin_factory=None):
        super(PiStopCrossing, self).__init__(
            up_down=_PiStopCrossingSegment(
                TrafficLights(4, 17, 18, pwm=pwm, pin_factory=pin_factory),
                LedBoard(red=25, green=24, pwm=pwm, pin_factory=pin_factory),
                Button(12, pin_factory=pin_factory),
                pin_factory=pin_factory
            ),
            left_right=_PiStopPedestrianCrossing(
                TrafficLights(27, 22, 23, pwm=pwm, pin_factory=pin_factory),
                LedBoard(red=5, green=6, pwm=pwm, pin_factory=pin_factory),
                Button(13, pin_factory=pin_factory),
                pin_factory=pin_factory
            ),
            buzzer=Buzzer(19, pin_factory=pin_factory),
            order=('up_down', 'left_right', 'buzzer'),
            pin_factory=pin_factory
        )
```
and then
```python
crossingboard = PiStopCrossing()
crossingboard.left_right.trafficlights.green.on()
crossingboard.up_down.trafficlights.amber.blink()
sleep(5)
crossingboard.left_right.off()
crossingboard.up_down.off()
```

> I can't see any example code

Yeah, doesn't seem to be anything relevant at https://github.com/4tronix?tab=repositories

Ping @4tronix in case they want to provide any extra info (e.g. are the buttons pull-up or pull-down?)

sbcshop, 2017-12-11:
> tab=repositories

Hi Andrew,

For this, only you need to change the file name of PiTraffic.py to any
other name like SBTraffic.py etc.

 Code is also attached for reference.

GAJENDER SINGH
DIRECTOR
<02035140914>02035140914
<#UNIQUE_ID_SafeHtmlFilter_UNIQUE_ID_SafeHtmlFilter_UNIQUE_ID_SafeHtmlFilter_UNIQUE_ID_SafeHtmlFilter_>
gajender.singh@sb-components.co.uk
https://shop.sb-components.co.uk
<https://twitter.com/sb_ltd> <https://www.facebook.com/sbcomponentsltd>
<https://www.youtube.com/channel/UC-HPa0xV9LtjqT79aajhGfQ>
<https://plus.google.com/115397217718246801767>
<https://shop.sb-components.co.uk>
Unit B, 58B Kingsbridge Crescent,
Southall,  UB1 2DL  United Kingdom
Vat: GB154489384

This electronic transmission is strictly confidential and intended solely
for the addressee. It may contain information which is covered by legal,
professional or other privilege. If you are not the intended addressee, you
must not disclose, copy or take any action in reliance of this
transmission. If you have received this transmission in error, please
delete it and notify us as soon as possible. The Internet cannot guarantee
the integrity of this message, SB Components Ltd will not therefore be
liable for the message if modified.

closed 2021-02-26 by bennuttall