gpiozero/gpiozero #609: Closer integration with pinout.xyz ?

lurch, 2017-11-30:
Looking at e.g. https://pinout.xyz/pinout/scorezero it just occurred to me how **awesome** it would be if GpioZero could take some of the data from the https://github.com/gadgetoid/Pinout.xyz repo and use that to auto-generate GpioZero python classes? :grin: 
E.g. if https://raw.githubusercontent.com/Gadgetoid/Pinout.xyz/master/src/en/overlay/scorezero.md was modified to specify that it should create a ButtonBoard, then GpioZero could somehow auto-generate:
```python
class ScoreZero(ButtonBoard):
    """
    Could have some auto-generated description here, similar to https://pinout.xyz/pinout/scorezero
    """
    def __init__(self, pin_factory=None):
        super(ScoreZero, self).__init__(
            start=5,
            select=6,
            up=22,
            right=23,
            down=24,
            left=25,
            b=26,
            a=27,
            order=('start', 'select', 'up', 'right', 'down', 'left', 'b', a'),
            pin_factory=pin_factory
        )
```

ping @bennuttall @Gadgetoid 
Perhaps this also links in with #548 , but without the 'connections' stuff?

@Gadgetoid And on a related note, perhaps you could extend pinout.xyz so that for those boards fully supported by GpioZero, you add a link to their GpioZero class docs? :-)
E.g. https://pinout.xyz/pinout/traffic_hat could have a link to https://gpiozero.readthedocs.io/en/stable/api_boards.html#traffic-hat

Gadgetoid, 2017-11-30:
This is a bonkers idea, but it seems possible. I'm definitely keen to make as much API functionality available as I can for the next phase of Pinout growth. One of the key ideas behind bringing boards to Pinout was always to aggregate data so we could use it for the greater good. Anything that achieves that goal is awesome!

lurch, 2017-11-30:
BTW @Gadgetoid I just happened to spot that https://pinout.xyz/pinout/rtk_motor_controller _does_ have a  link to its GpioZero docs (cool!), but it'd be better if it linked to http://gpiozero.readthedocs.io/en/stable/api_boards.html#ryanteck-mcb-robot rather than http://gpiozero.readthedocs.io/en/v1.3.1/api_boards.html#ryanteck-mcb-robot

lurch, 2017-11-30:
At the very least, the lovingly curated pinout data on https://pinout.xyz will make it easier for Python developers to hand-write the corresponding GpioZero classes :)

bennuttall, 2017-12-01:
> This is a bonkers idea, but it seems possible.

I want those to be my last words.

bennuttall, 2017-12-01:
Regarding a `ScoreZero` implementation, I started working on one the other night. but my idea required adding a `DPad` class for the up/down/left/right, and then required `ButtonBoard` to be nestable, hence #607, rather than just a linear `ButtonBoard`. Any thoughts?

lurch, 2017-12-01:
Without nestable ButtonBoard classes, I guess you could group the buttons by doing:
```python
class ScoreZero(CompositeDevice):
    def __init__(self, pin_factory=None):
        super(ScoreZero, self).__init__(
            dpad=ButtonBoard(
                up=22,
                right=23,
                down=24,
                left=25,
                order=('up', 'right', 'down', left'),
                pin_factory=pin_factory
            ),
            buttons=ButtonBoard(
                start=5,
                select=6,
                b=26,
                a=27,
                order=('start', 'select', 'a', 'b'),
                pin_factory=pin_factory
            ),
            order=('dpad', 'buttons'),
            pin_factory=pin_factory
        )
```

closed 2019-02-01 by bennuttall