gpiozero/gpiozero #610: Update remote GPIO docs to include new SD card-less Pi Zero OTG USB boot mode

bennuttall, 2017-12-01:
Today's Stretch-based release of Raspberry Pi Desktop x86 includes a new tool which allows you to boot a Pi Zero over USB, without an SD card, and use the GPIO pins on the Pi. This works by booting a mini version of Raspbian in RAM and running pigpiod.

This means that GPIO Zero works out of the box. You just need to launch Python or an IDE with the pin factory set (not essential due to fallback), and the IP address of the Pi, which is always the same (can be enumerated for multiple Pis but this doesn't seem to work at present).

This is a killer feature! It's always been possible to do this but you needed an SD card, with some config changes, and you needed to remember to run pigpiod. Bit of a faff. But now it's plug-and-play and there's no SD card to worry about.

dglaude, 2017-12-02:
Does the Raspberry Pi Foundation released similar tools for Windows and Mac?
I mean Raspberry Pi x86 Desktop is not the most common thing (even if I have one at home).

waveform80, 2018-02-20:
@dglaude not currently and while I can't speak for the Pi Foundation, I wouldn't hold my breath on that - it's a fairly tall order (there's no common API for detecting insertion of hardware across platforms, there's plenty of variation in setting up USB gadgets across platforms, etc. - in other words it's not an "easy" port).

commit 8b3e872b (2018-02-20): Add docs for Pi Zero OTG using USB boot GUI, close #610
closed 2018-02-20 by bennuttall