gpiozero/gpiozero #611: Is there a random_values that sets pwm leds to 0.0-1.0 or could this be added?

sjd4, 2017-12-01:
I got the 3D X-mas tree from PiHut and wanted some lights to turn of temporarily.  random_values() never gave a 0 for led.source.  I looked the source for random_values and created a different solution:
`from random import randint`
`...`
`while True:`
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;`yield radint(0,10)/10`

Is there another way to do this or could this be added?

lurch, 2017-12-03:
I think https://gpiozero.readthedocs.io/en/stable/api_tools.html#gpiozero.tools.quantized might do what you're after?

sjd4, 2017-12-04:
What would I supply for values?  There is no other input device.  Just outputting values to LEDs.  I wanted to create a generator that outputs random numbers 0.0 – 1.0 including both 0.0 and 1.0.  led.source takes a generator, but I think quantized needs some kind of input set under values in the constructor?

def quantized(values, steps, input_min=0, input_max=1):

lurch, 2017-12-05:
Well, technically the `random_values` _does_ output both 0.0 and 1.0 - it's just _very_ unlikely to output those exact values, much as it's very unlikely to output exactly 0.50000000.

The sourcetools are designed so that they can be 'chained' together (see https://gpiozero.readthedocs.io/en/stable/source_values.html , although I now see it doesn't show chaining multiple sourcetools @bennuttall ). In your case you'd want to do:
```
led.source = quantized(random_values(), 10)
```
which should give you a random value from the list (0.0, 0.1, 0.2, 0.3, 0.4, 0.5, 0.6, 0.7, 0.8, 0.9, 1.0)
Although I've not tested this myself, so I can't say for certain whether both 0.0 and 1.0 will **actually** be included in the output.

@waveform80 I wonder if it'd be worth add a `quantized_random_values` sourcetool that better provides the behaviour that @sjd4 is expecting?

bennuttall, 2017-12-05:
@lurch Yes, that works!

sjd4, 2017-12-06:
@lurch  It worked for me too!  This works fine.  If the devs think it's worth making a quantized_random_values() the feature gets my vote.  Either way this issue is resolved.

lurch, 2017-12-06:
Oh cool! I wasn't sure if it'd round-down (in which case 1.0 wouldn't be included in the output) or if it would round-up (in which case 0.0 wouldn't be included) - but if it works as expected, that's great to hear :-)

@bennuttall Maybe this'd make a good example for the docs?

sjd4, 2017-12-07:
I just checked and it does round-down.  It never outputs 1.0.

It would definitely make a good example for the docs.  I didn't understand that the values variable in the quantized constructor was a generator.

normanr, 2018-04-22:
btw, you can get quantized to include 1.0, by changing the quantized call like such: instead of ```quantized(random_values(), n)``` use ```quantized(random_values(), n-1, input_max=1-1/n)```.  This makes it such that the 1.0 bucket has an equal chance of being selected as the other ones.