gpiozero/gpiozero #612: /var/log filled sd card after using remote gpio library

av4625, 2017-12-03:
I wrote this script (I may be using it wrong)

[image]

But it has seemed to fill my sd card with massive logs in /var/log
daemon.log and syslog and syslog.1 are huge and contain the following line:
python3[1793]: ALSA lib pcm.c:7843:(snd_pcm_recover) overrun occurred

lurch, 2017-12-03:
Why was this closed? ;)
What version of GpioZero are you using? What version of Raspbian are you using?

av4625, 2017-12-03:
This was the only thing I had done on the raspberry pi recently so thought it was the issue.
I also have google aiy on the raspberry pi and I think it may be the issue as it just randomly stopped working and I cant get it to work again.
Gpiozero 1.4.0 and Jessie

lurch, 2017-12-04:
I don't know, that's not a combination I've tried myself. If you can recreate this problem on a freshly-written SD card then please re-open this issue.

av4625, 2017-12-11:
I just got this again with a freshly written SD card, daemon.log and syslog got filled agin.

This time it uses the newer version of google aiy which is different than the old version.

This is everything that is installed.
MySQL
sudo apt-get install mysql-server --fix-missing
sudo apt-get install mysql-client php5-mysql

Connect to MySQL from python
sudo apt-get install python-dev libmysqlclient-dev
sudo apt-get install python-mysqldb
pip install mysqlclient

Vim
sudo apt-get install vim

Apache
sudo apt-get install apache2

PHP
sudo apt-get install php5cd /var

The remote daemon is also on

lurch, 2017-12-12:
Hmmm, then perhaps there's some kind of conflict with google AIY and pigpiod? (pigpiod is the underlying library that GpioZero uses for remote gpio)

Might be worth chasing up this bug on each of those projects' websites.
https://aiyprojects.withgoogle.com/
http://abyz.me.uk/rpi/pigpio/pigpiod.html

(ping @joan2937 )

bennuttall, 2017-12-12:
Yeah this isn't a bug in gpiozero, it's something going on with pigpio.

joan2937, 2017-12-12:
The logs are apparently being created by the ALSA library.  By default pigpio uses the PCM peripheral to time the GPIO sampling.  Presumably Google AIY is also using the PCM peripheral via the ALSA library.

One way to test would be to start the pigpio daemon with the **-t0** option (see http://abyz.me.uk/rpi/pigpio/pigpiod.html).  pigpiod will then use the PWM peripheral instead of the PCM peripheral. 

E.g. try something like the following.

```
sudo killall pigpiod
sudo pigpiod -t0
```

lurch, 2017-12-12:
@joan2937 Oooh, interesting. As somebody who knows very little about all this, does using PCM over PWM (or vice versa) allow more precision in pigpiod, or are they both equally capable?
Would there be a possibility to also have an additional `-t 2` clock option, to then have pigpiod use **No** clock peripheral? (with errors then thrown as appropriate if the user tries to use a function which requires a clock peripheral)   Or perhaps that's a stupid question? ;-)

closed 2017-12-12 by bennuttall

joan2937, 2017-12-12:
@lurch 

PWM and PCM are equal as far as the timing is concerned.

They are actually used to pace the DMA transfers (an idea of Gerts, first implemented by Richard Hirst and used in his servoblaster).  The GPIO are sampled by a DMA transfer from the GPIO register to memory.

High speed sampling and accurate time-stamping are central to pigpio.  They are the reason it exists.  This would be impossible without using DMA transfers.  As a byproduct you also get pretty accurate PWM generation on all the GPIO for free.

av4625, 2017-12-12:
Thanks guys for all your help, (sorry for posting the issue in the wrong place, I was confused at first when it happened)

I have tried joan's suggestion and it seems to work so far.
It throws this error:
/home/pi/AIY-voice-kit-python/src/aiy/_drivers/_led.py:51: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.

This is the led for the button but the led seems to work

lurch, 2017-12-13:
@joan2937 As far as GpioZero is concerned, the remote-connection feature is pigpio's killer feature ;-)
I don't think GpioZero actually makes use of High speed sampling and accurate time-stamping (although I'm sure other people must find them useful) - GpioZero kinda adopts a lowest-common-denominator approach, so that different GPIO libraries can be swapped in and out.

@av4625 Glad to hear that things are working now :-)

joan2937, 2017-12-13:
@lurch 

One thing I would do, time permitting, is copy the source base and make a new project.  I'd then strip out all the DMA sampling, waves, PWM.  That would leave generic code which should work on any Linux platform.  There would still be a lot of functionality left.  Sampling and PWM could be added back in albeit in a degraded pure software form.