gpiozero/gpiozero #613: LEDBoard close() method disables the defined Button definition or handler.

tarapitha, 2017-12-11:
The handler 'nxt_mode' is called as expected on following Python3 code snippet:
---
tree = LEDBoard( *range(4,28),pwm=True)
button = Button(3)
button.when_pressed = nxt_mode
---
but when the LEDBoard object is closed after the button definition, then the button handler fails to be called:
---
tree = LEDBoard( *range(4,28),pwm=True)
button = Button(3)
button.when_pressed = nxt_mode
# This close will void the button handler or button definition and pressing GPIO3 button has no effect.
tree.close()
tree = LEDBoard( *range(4,28),pwm=True)
---

bennuttall, 2017-12-12:
Wow. That's unusual. I have confirmed, and limited to `RPiGPIOFactory`.

If you use `MockFactory` or `PiGPIOFactory`, it works as expected.

Run `sudo pigpiod` and then launch with `GPIOZERO_PIN_FACTORY=pigpio python3` (or idle or whatever) and it works fine.

lurch, 2017-12-12:
There's so many [open issues](https://sourceforge.net/p/raspberry-gpio-python/tickets/) on RPi.GPIO that it's hard to know if any of them already cover this particular bug.

bennuttall, 2017-12-12:
I have a feeling it's a bug in our pin implementation for RPi.GPIO rather than a big in RPi.GPIO, but it could be.

waveform80, 2017-12-12:
Definitely an RPi.GPIO bug. Dashing to write this so forgive the brevity: event_cleanup() in event_gpio.c has a bug: it always cleans up all events not just the one it's called for due to missing {braces}. Looks like [this ticket](https://sourceforge.net/p/raspberry-gpio-python/tickets/145/) *might* fix it but I haven't had time to check it (my hunch is it doesn't because there's no longer anything advancing the g pointer when the match isn't true).

Anyway, must dash - hope that's enough to go on for now!

bcroston, 2017-12-12:
Added a fix to RPi.GPIO as per the missing braces ticket mentioned by @waveform80 above.  To install this development version of RPi.GPIO using pip (assuming you are working in a virtual environment):
$(venv) sudo apt-get install mercurial
$(venv) pip install hg+http://hg.code.sf.net/p/raspberry-gpio-python/code#egg=RPi.GPIO

Does it fix this gpiozero problem?  If so, I will upissue RPi.GPIO.

bennuttall, 2017-12-13:
That fixed it for me! Thanks Ben

tarapitha, 2017-12-13:
I can confirm, this fixes the mentioned bug. Thanks, that was fast!

lurch, 2017-12-14:
Should this issue be re-opened until RPi.GPIO 0.6.4 is released and included in the Raspbian update-repos?

tarapitha, 2017-12-14:
Maybe. I am new to this thing and closed it actually by accident... My  intention was to send the comment, but clicked the wrong button. It might be justified to reopen it, but I'm not in hurry to click the buttons again. Can anyone reopen the issue?

bennuttall, 2018-02-24:
@bcroston when do you plan to release 0.6.4?

bennuttall, 2018-09-24:
I'm closing this again as it's not a bug in gpiozero. Hopefully rpi.gpio 0.6.4 will be released at some point.

closed 2018-09-24 by bennuttall

bcroston, 2018-10-26:
RPi.GPIO 0.6.4 is now released.