gpiozero/gpiozero #615: SmoothedInputDevice takes Median (instead of Average) of queued values

holzfigure, 2018-01-21:
While in the documentation it says that the mean or average of the values in the internal queue will be compared to "threshold" to determine whether device "is_active", value returns the median of the queue. I fixed it for myself in "mixins.py", by importing the mean and then passing it as the average keyword parameter to the __init__-function of GPIOQueue.

I wonder whether the taking median serves a purpose i'm unaware of (apart from contradicting the documentation, hehe), but given digital inputs, the median can only return 0 or 1 (or very rarely 0.5), rendering the threshold also a bit useless (i mean it does work, strictly speaking).

I suggest changing that behavior?

lurch, 2018-01-21:
Oh, well spotted! #615 should (hopefully) fix this - if you can test it that'd be great.

holzfigure, 2018-01-23:
just tested it, works fine. It's actually easy to test: just connect two GPIO pins, set one to `PWMOutputDevice`, set it's value to something between 0 and 1. Then use the second GPIO pin as `SmoothedInputDevice` and read in the values,)

lurch, 2018-02-21:
I guess this can be closed now?

closed 2018-02-21 by holzfigure