gpiozero/gpiozero #621: Bitbucket URL in Makefile doesn't work

lurch, 2018-02-21:
In the "Horrid hack" section in the `Makefile`, there's a URL which no longer exists https://bitbucket.org/pypa/setuptools/raw/bootstrap/ez_setup.py
(and it looks like it hasn't existed for [almost 2 years](https://bitbucket.org/pypa/setuptools/commits/all) ?)

I guess either this URL should be updated to some other location that _does_ exist, or the "Horrid hack" section should be updated to do something else?

bennuttall, 2018-02-24:
What's the consequence of this 404?

SteveAmor, 2018-03-27:
I guess @waveform80 will know what the consequence of the 404 is?
The file is preserved here:
https://bootstrap.pypa.io/ez_setup.py
It has been deprecated.  More information here https://github.com/pypa/setuptools/issues/581
It may already be picked up by the following (making the "Horrid hack" redundant)?
https://github.com/RPi-Distro/python-gpiozero/blob/5032124b6de5f09b8612722b51f965ffd9dce298/Makefile#L104

bennuttall, 2019-09-20:
#799 should mean we can remove the broken link :)