gpiozero/gpiozero #624: Cannot import name LED

consolacion, 2018-03-01:
Trying some of the simplest gpiozero examples like:
```
from gpiozero import LED
from time import sleep

led = LED(17)

while True:
    led.on()
    sleep(1)
    led.off()
    sleep(1)
```

and
```
from gpiozero import LED, Button
from signal import pause

led = LED(17)
button = Button(3)

button.when_pressed = led.on
button.when_released = led.off

pause()

```
But keep getting the "Cannot import name LED" error.
Using Raspian Stretch with python 2.7.
gpiozero lib is version 1.4.1

Have no idea what is wrong. Other 'can't import name XXXX' errors didnt really help me further.
Anyone any suggestions?

bennuttall, 2018-03-01:
You have named your file `gpiozero.py` and the import line is trying to import LED from that file, not from the library.

You need to delete `gpiozero.py` (and if there's a `gpiozero.pyc` delete that too).

consolacion, 2018-03-01:
You sir are either a mind reader or have secret access to my drive. Indeed that seems to be the issue. I presume that same issue  counts for any file name with gpiozero in it, even gpiozero_led.
thank you!!

bennuttall, 2018-03-01:
Hehe I know, it's a common problem.

> I presume that same issue counts for any file name with gpiozero in it, even gpiozero_led.

No - only `gpiozero.py` will cause this error.

closed 2018-03-01 by bennuttall

consolacion, 2018-03-01:
Tnx again. Seems  a great and easy library