gpiozero/gpiozero #628: Alternative ButtonBoard

bennuttall, 2018-03-21:
Currently the `ButtonBoard` class has an n-tuple value:

```python
>>> bb = ButtonBoard(2, 3, 4, 5)
>>> tuple(bb.value)
(0, 0, 0, 0)
```

and if one button is pressed, the activated callback is fired:

```python
>>> bb.when_pressed = lambda: print("Pressed")
>>> bb.all[0].pin.drive_low()
Pressed
>>> tuple(bb.value)
(1, 0, 0, 0)
```

I wondered if we could introduce a threshold, so you could specify how many active buttons were required for the event to fire, e.g:

```python
>>> bb = ButtonBoard(2, 3, 4, 5, threshold=4)
```

i.e. all four buttons must be pressed for the event to fire.

@waveform80 suggested that this should be a separate class - similar to the differentiation between `LEDBoard` and `LEDBarGraph` where the new class would have a single 0-1 value rather than an n-tuple. So perhaps `ButtonGroup`?

This would only allow either one-button activations (with `ButtonBoard`) or all-button activations (with `ButtonGroup`) but nothing in-between (unless this can be solved with nesting mentioned in #607) - not a huge problem as I don't expect this to be too common (and can be achieved using other methods).

lurch, 2018-03-22:
What's the imagined use-case for having a ButtonBoard that only fires it's `when_pressed` event when e.g. any 3 out of a possible 5 buttons are pressed? (if I'm understanding your `threshold` suggestion correctly)

Or are you specifically only asking for an event that fires when _all_ buttons are pressed. Would adding an `when_all_pressed` event to the existing ButtonBoard class solve your requirement?

bennuttall, 2019-03-24:
Closing this in favour of #522 - moving this discussion there

closed 2019-03-24 by bennuttall