gpiozero/gpiozero #630: Development - mkvirtualenv: command not found

SteveAmor, 2018-03-25:
In the docs, section 11.1. (Development installation) details the process to install a virtual Python environment.  However, after installing the Python packages/modules, the command ```mkvirtualenv -p /usr/bin/python3 python-gpiozero``` returns ```-bash: mkvirtualenv: command not found```.  To get around this, I uninstalled virtualenvwrapper python-virtualenv python3-virtualenv and used pip to install what I could to "get it working".  I'm no expert with this otherwise I'd raise a PR with a suggested fix.
Tested with both apt and apt-get on new install of Stretch Lite (after a sudo apt-get update).

SteveAmor, 2018-03-26:
Just been looking at piwheels and the test and the virtualenv it uses so I'm working on a fix for sectionn 11.1.  Will raise a PR later (I hope).

SteveAmor, 2018-03-26:
Fixed in #632

SteveAmor, 2018-03-27:
According to the official documentation (https://virtualenvwrapper.readthedocs.io/en/latest/install.html) after creating a virtualenv directory using ```mkdir ~/.virtualenvs``` the following should be added to .bashrc:
```
export WORKON_HOME=/home/pi/.virtualenvs
source /usr/share/virtualenvwrapper/virtualenvwrapper.sh
````
Then run ```source ~/.bashrc```

waveform80, 2019-02-10:
Fixed in 3685ffb72a1ffb70bc18a1fd62169b442adf0999 - remember to close when docs-tidy-up PR is merged

closed 2019-02-10 by bennuttall