gpiozero/gpiozero #639: Add fade method to PWMLED, RGBLED and LEDBoard

bennuttall, 2018-04-05:
As suggested in #449, we should consider adding a `fade` method to PWMLED and RGBLED so that users can fade from one value to another in a specified amount of time:

```python
led.fade(start=0, end=1, fade_time=10) # example of named args but these would probably be the defaults
```

The effect would be roughly equivalent to settling the source to a list values stepping from one value to the other:

```python
led.source_delay = 0.1
led.source = [i/100 for i in range(100)]
```

However it would be implemented like `blink` and `pulse` rather than by setting source.

Request for comments on the API!

lurch, 2018-04-06:
I'm obviously a bit biased, but why not the API that I suggested?
`led.fade_to(value, fade_time=1, background=True)` :man_shrugging: 

* The `fade_time=1` and `background=True` defaults match up better with the existing `blink` and `pulse` API
* I think it makes sense for the 'start' of the fade to always be the LED's _current_ value? (especially useful if doing a sequence of fades with RGBLED)

From a technical point of view, it probably makes sense to add `.fade_to` (or `.fade`) to every class that currently implements `.pulse`, i.e. `PWMOutputDevice`, `RGBLED` and `LEDBoard`.

tinitaet, 2021-10-17:
Better late than never:
I  was annoyed by this as well, so I created a new pull request with this feature for all PWMDevices. 
@lurch I took the middle ground for the start value by defaulting to the current value with the option to set it manually.