gpiozero/gpiozero #640: Add connect method to devices with source

bennuttall, 2018-04-05:
I really like how simple [source/values](https://gpiozero.readthedocs.io/en/stable/source_values.html) makes connecting up devices in one line:

```python
led.source = btn.values
```

However, a combination of the magic behind it and the naming convention makes me consider it an advanced feature I wouldn't really teach to kids or beginners.

If we had a way that was as simple as this, but seemed more beginner friendly and approachable, that would be great. This is oddly something I considered in a gpiozero pre-release (before source/values), but something I've reconsidered lately. Here's what I have in mind:

```python
led.connect(btn)
```

which would set the LED's source to the button's values, as above. It could be implemented in `SourceMixin`. We could possibly inspect the "shape" of the value of the two devices to make sure they're compatible.

lurch, 2018-04-06:
As I mentioned the other evening - my only concern with this would be how does a newcomer to GpioZero know whether to do:
```
led.connect(btn)
```
(which I guess could be 'read' as "led is on when button is pressed")
or
```
btn.connect(led)
```
(which I guess could be 'read' as "when button is pressed, turn on led")
?

Hmm, what if (just off the top of my head) instead of `.connect` we had `.controls` and `.controlled_by` ?
So that `led.controlled_by(btn)` and `btn.controls(led)` would both have the same effect as `led.source = btn.values` ? Or maybe that's too much API bloat... \*shrug\*

Or perhaps another solution to the problem might be some property aliases - do you think
```
led.input = btn.output
```
would be any more beginner-friendly @bennuttall ?

[disclaimer: I've not looked at any of the GpioZero code for quite a while...]

ukBaz, 2018-04-06:
These seem like some good ideas to me. I particularly like the `led.input = btn.output`.
Then my preference would be `led.controlled_by(btn)` and `btn.controls(led)`

bennuttall, 2018-04-09:
Alternatively, `.controls` could be defined in `SourceMixin` as `self.source = other.values` and in `ValuesMixin` as `other.source = self.values` ¯\_(ツ)_/¯

lurch, 2018-04-09:
I guess `ValuesMixin` would have `.controls` and `SourceMixin` would have `.controlled_by` ?

I think it'd be too confusing if you had a method named `.controls` in both `SourceMixin` and `ValuesMixin` ?

bennuttall, 2018-11-26:
@basak suggested something and I thought it was a better solution to this suggestion! With this method you could do:

```python
led.source = btn
led.source = negated(btn)
led.source = all_values(btn1, btn2)
motor.source = scaled(pot, -1, 1)
led.source = booleanized(pot, 0.25, 0.75)
led.source = multiplied(pot1, pot2, pot3)
garden.source = all_values(booleanized(light, 0, 0.1), motion)
```

instead of:

```python
led.source = btn.values
led.source = negated(btn.values)
led.source = all_values(btn1.values, btn2.values)
motor.source = scaled(pot.values, -1, 1)
led.source = booleanized(pot.values, 0.25, 0.75)
led.source = multiplied(pot1.values, pot2.values, pot3.values)
garden.source = all_values(booleanized(light.values, 0, 0.1), motion.values)
```

PR #698

bennuttall, 2018-11-26:
Unless anyone can think of a reason not to, this will be in the next release. @lurch @ukBaz any thoughts on the above comment / #698?

commit 96cf4038 (2019-01-14): Merge pull request #698 from RPi-Distro/connect-devices

Allow source tools to take device object as well as device.values or just values, close #640
closed 2019-01-14 by bennuttall