gpiozero/gpiozero #644: DistanceSensor returns Zero when distance less than 25cm

AmirSasson, 2018-04-18:
i have checked and the hardware is fine i am using HCSR04  on RPI Zero W
it seems like the echo is returning "too fast" and results zero when the distance less than ~25cm
i know there were many discussions on this, but all the issues are closed,
i am using the latest release 1.4.1

bennuttall, 2018-04-18:
This issue was fixed in 1.4.1 but there's a chance you're doing something different which is causing a replication of it.

Can you confirm:

- You are using Raspbian Stretch
- You have not installed gpiozero with pip
- You are definitely using v1.4.1 ([please check](https://gpiozero.readthedocs.io/en/stable/faq.html#how-can-i-tell-what-version-of-gpiozero-i-have-installed))
- You are using the RPi.GPIO Pin Factory (please check by running):
  ```python
  from gpiozero import Device
  print(Device.pin_factory)
  ```
- You have tested the most basic version of the DistanceSensor code (nothing else could be slowing it down)

SteveAmor, 2018-06-22:
This is the behaviour that I have seen for a PiZero as I reported in https://github.com/RPi-Distro/python-gpiozero/pull/595#issue-147878485 which is identical code which subsequently ended up in V1.4.1.
The more accurate fix would be as @waveform80 states at the end of https://github.com/RPi-Distro/python-gpiozero/issues/385#issuecomment-227716420

bennuttall, 2019-01-25:
This *should* be fixed on master now. Two big improvements to the way we calculate distance thanks to great effort from @waveform80. Please test if you can:

```
git clone https://github.com/RPi-Distro/python-gpiozero
cd python-gpiozero
sudo pip3 install .
```

The v1.5.0 release is expected very soon.

closed 2019-02-01 by bennuttall