gpiozero/gpiozero #650: Can the SPI bus speed be changed?

tinue, 2018-05-26:
To get independent from an external library, I am testing the GPIO Zero library for my [APA102 driver](https://github.com/tinue/APA102_Pi). Currently, I am using the [Adafruit SPI](https://github.com/adafruit/Adafruit_Python_GPIO) implementation. Initially I used [py-spidev](https://github.com/doceme/py-spidev) which only supports hardware SPI.

When I use GPIO Zero over the hardware SPI, it paints much slower (by a factor of at least 10, possibly much more) compared to both of the libraries mentioned before. With the other libraries, I have to crank up the bus speed to 8MHz to get good results, though.

So I wonder: Can this be accomplished with GPIO Zero as well?

Software SPI (bitbang) is another matter altogether: The Adafruit library is some orders of magnitude faster than GPIO Zero. I suspect that they use C code for the actual bit banging, but I have not looked at the source code to verify this. For me, it would not be too much of an issue, because the software SPI devices tend to have few LEDs (Pimoroni Blinkt! for example with 8 LEDs).

P.S. I am aware of [Issue 551](https://github.com/RPi-Distro/python-gpiozero/issues/551)

bennuttall, 2018-05-29:
SPI speed is dependent on the pin factory used. You probably want to switch to pigpio: https://gpiozero.readthedocs.io/en/stable/api_pins.html#changing-the-pin-factory

Alternatively try the pigpio C library itself: http://abyz.me.uk/rpi/pigpio/examples.html#Python_test-APA102_py

tinue, 2018-06-03:
Thank you for the hint! I will definitly have a look at the pigpio library.

tinue, 2018-11-25:
I did try out pigpio, but I still have the same problem. It is extremely slow, even if hardware SPI is used. I see that the bus speed is hardcoded and can't be changed. For example, in class PiGPIOHardwareSPI, line 320: self._baud = 500000

I tried to overwrite the variable after I retrieved the class, but apparently it's too late at this point, because SPI is already initialized. I would have to either have a setter that re-initializes the SPI, or I would have to pass the bus speed in the constructor.

What do you think?

jbshaler, 2019-05-24:
I had the same problem. As a workaround, after setting `_baud` to a higher value, you can use one of the `PiGPIOHardwareSPI` property setters that automatically closes and re-opens the connection. This is working for me:
    
    spi = factory.spi(port=port, device=device)
    spi._baud = int(10e6)  # 10 MHz
    spi.clock_mode = spi.clock_mode  # automatically close and re-open with new baud rate

It seems like baud could be added as a property to `PiGPIOHardwareSPI`, with an implementation similar to `_set_clock_mode` and `_get_clock_mode`. Perhaps it should also be added to the `SPI` base class and `PiGPIOSoftwareSPI` class.

tinue, 2019-05-25:
Thanks! One question: Which factory are you using? This is my code:
`factory = PiGPIOFactory(host='localhost')`

While I can see a tiny bit of improvement, it's not a lot. Is there a factory that goes directly to the hardware, instead of to the daemon?

bennuttall, 2019-05-25:
No, that's just how pigpio works.

jbshaler, 2019-05-28:
@tinue I'm actually doing remote SPI using `factory = PiGPIOFactoy(host='169.254.225.2')`. I didn't actually measure how much my throughput improved with baud rate (just looked at the SCLK frequency), so maybe I'm not getting much benefit either.

joan2937, 2019-07-17:
I used to have a note in the pigpio Python SPI and I2C implementations suggesting that people use spidev for SPI and smbus for I2C **unless** they planned on using remote access.  There is a heavy overhead in using a client/server model.  As you have discovered if you do use SPI/I2C via pigpio then make each transaction as large as possible to minimise message overheads.

waveform80, 2021-03-12:
Fixed in b30983a1aeec9d83af555394c663c965761ac4b6

closed 2021-03-12 by waveform80