gpiozero/gpiozero #656: Can't Locate 'pinout' After Install 'pyhton-gpiozero'

blogabe, 2018-06-20:
I have an original RPi that I'm running with DietPi.  After installing python-gpiozero per the Instructions, I cannot execute or find the pinout command.

Is this an issue due to running on original RPi, DietPi, or something alltogether different?  Appreciate your help.

bennuttall, 2018-08-22:
Sorry I didn't see this issue when you posted it.

I think the `pinout` tool is only installed when you install the Python 3 version of the package. We can't bundle it with both versions as they would overlap, so we chose the Python 3 version.

```bash
sudo apt install python3-gpiozero
```

will bring the `pinout` tool in. Can you confirm?

blogabe, 2018-09-04:
Hi there.  Thanks.  I'm trying to remember if I tried that or not prior to posting, but can't recall.  I'll try it soon and will test it out. Thanks.

blogabe, 2018-09-13:
Can confirm this works now.  Thanks so much @bennuttall

closed 2018-09-13 by bennuttall