gpiozero/gpiozero #657: Request to add support to play audio file

winkleink, 2018-06-21:
For interactive projects it would be great to be able to play an audio file. wav/mp3/ogg. 
Maybe with volume control. 

Thinking for beginners it adds an alternative output to make learning more fun.

SteveAmor, 2018-06-22:
@winkleink how do you see this from a command point of view?  Something like:

from gpiozero import PlayWAV, Button
from system import pause

mysound = PlayWAV("/home/pi/tada.wav")
button = Button(21)

button.when_pressed = mysound.on  # ```on``` is consistent with gpiozero but ```play``` may be better

pause()

SteveAmor, 2018-06-22:
WAV files would be easier to handle as the system wav player ```aplay``` comes as standard with Stretch and Stretch Lite.  So we can make a system call to ```aplay```.  No way to adjust to volume with ```aplay``` though.  It would probably reside in other_devices.py ??

winkleink, 2018-06-22:
That would work.
Thought process is to reduce the number of import statements a beginner has to use.

bennuttall, 2018-06-22:
We wouldn't add this to gpiozero but instead recommend playing sounds with pygame:

```python
from gpiozero import Button
import pygame.mixer
from pygame.mixer import Sound
from signal import pause

pygame.mixer.init()

btn = Button(2)
drum = Sound("drum.wav")

btn.when_pressed = drum.play

pause()
```

One of the recipes in the docs includes a sound player for multiple sounds controlled by a series of buttons: https://gpiozero.readthedocs.io/en/stable/recipes.html#gpio-music-box

closed 2018-06-22 by bennuttall