gpiozero/gpiozero #659: How to handle loss of remote gpio

jacobthetechy, 2018-07-10:
Im having a issue with handling an remote led when the remote rPi is power cycled.

setup
```
from gpiozero.pins.pigpio import PiGPIOFactory
from gpiozero import LED

factory = PiGPIOFactory(host='10.96.36.183')
led = LED(17, pin_factory=factory)
led.on()
```
led turns on as expected. reboot 10.96.36.183

```
led.on()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/chem_sbtn/venv/lib/python2.7/site-packages/gpiozero/output_devices.py", line 159, in on
    self._write(True)
  File "/home/pi/chem_sbtn/venv/lib/python2.7/site-packages/gpiozero/output_devices.py", line 60, in _write
    self.pin.state = self._value_to_state(value)
  File "/home/pi/chem_sbtn/venv/lib/python2.7/site-packages/gpiozero/pins/__init__.py", line 247, in <lambda>
    lambda self, value: self._set_state(value),
  File "/home/pi/chem_sbtn/venv/lib/python2.7/site-packages/gpiozero/pins/pigpio.py", line 219, in _set_state
    elif self.function == 'input':
  File "/home/pi/chem_sbtn/venv/lib/python2.7/site-packages/gpiozero/pins/__init__.py", line 225, in <lambda>
    lambda self: self._get_function(),
  File "/home/pi/chem_sbtn/venv/lib/python2.7/site-packages/gpiozero/pins/pigpio.py", line 192, in _get_function
    return self.GPIO_FUNCTION_NAMES[self.factory.connection.get_mode(self.number)]
  File "/home/pi/chem_sbtn/venv/lib/python2.7/site-packages/pigpio.py", line 1322, in get_mode
    return _u2i(_pigpio_command(self.sl, _PI_CMD_MODEG, gpio, 0))
  File "/home/pi/chem_sbtn/venv/lib/python2.7/site-packages/pigpio.py", line 984, in _pigpio_command
    dummy, res = struct.unpack('12sI', sl.s.recv(_SOCK_CMD_LEN))
struct.error: unpack requires a string argument of length 16
```
try to close the pin to reinitialize.

`led.close()`

the control raspberry pi python terminal hangs.

So how should this be handled when the remote pi gets power cycled?

bennuttall, 2018-07-11:
Unfortunately the pigpio Python client is not resilient to connection failures. See #516.

However you can get around this by regularly recreating the pin factory connection. For example something like this:

```python
while True:
    factory = None
    while not factory:
        try:
            factory = PiGPIOFactory('10.96.36.183')
        except OSError:
            sleep(1)
    led = LED(17, pin_factory=factory)
    led.on()
    sleep(60)
```

jacobthetechy, 2018-07-11:
@bennuttall Running the code does not work. This is what I get

`Exception struct.error: error('unpack requires a string argument of length 16',) in`

 then the interpreter hangs. Its passing the try:except with no issue but throws the error when reassigning led.

 Any other suggestions? Could this be a new feature to better handle this case?

Dreded, 2020-02-13:
well a year late but I was running into this issue too... so the solution is actually quite simple.
import struct
```
from gpiozero import LED
from gpiozero.pins.pigpio import PiGPIOFactory
from time import sleep

while True:
    factory = None
    while not factory:
        try:
            factory = PiGPIOFactory('10.96.36.183')
        except IOError:
            sleep(5)
    led = LED(17, pin_factory=factory)
    led.on()
    sleep(60)
    try:
        led.close()
        factory.close()
    except struct.error:
        continue
```

closed 2021-03-12 by bennuttall