gpiozero/gpiozero #664: gpiozero DistanceSensor pin_factory errors

greatworkscph, 2018-09-07:
Hi,
I get good Readings of the DistanceSensor class used on a pi, but haven't been enable to get Readings from an another Pi using DistanceSensor with pin_factory remotely.
Does anyone have experience with this?
Thanks

greatworkscph, 2018-09-07:
... And here is my code:
`import gpiozero
from gpiozero import DistanceSensor
from gpiozero.pins.pigpio import PiGPIOFactory
from time import sleep

factory = PiGPIOFactory(host='remotepi.local')

# Sensors
sensor = DistanceSensor( echo=14, trigger=15, max_distance=3.0, threshold_distance=2.0, pin_factory=factory)

while True:
    print('Distance: ', sensor.distance * 100)
    sleep(1)`

The sensor gives good readings on the pi where it's wired but not remotely...

waveform80, 2018-09-13:
Yes - in the current master branch, the pin timing (which is critical for DistanceSensor) will be *awful* when used with the pigpio implementation. I'll stress this is our fault and not pigpio's which actually has incredibly good timing but my first attempt at making the pin drivers all compatible basically ignored pigpio's extra timing information (because it was quicker than re-working the underlying pin API).

Anyway, the good news is we already have a fix but it isn't released yet. My [pin-event-timing branch](https://github.com/waveform-computing/gpio-zero/tree/pin-event-timing) should clear all this up and make the DistanceSensor much more accurate remotely (and a bit more accurate locally too). However, it's not *quite* ready for PR yet - needs a bit more testing. So far I've only had the time to test it with the distance sensor and because this alters the pins API in a few fundamental ways I really need to sit down and test it with *lots* of things. That said, if you fancy giving it a whirl yourself, do feel free to do so. Just follow the [development](https://gpiozero.readthedocs.io/en/stable/development.html) chapter in the docs, but clone my fork of the repo (github.com/waveform80/gpio-zero) and checkout the pin-event-timing branch afterward.

greatworkscph, 2018-09-17:
Fantastic!
Thanks for your help!

commit 415568c3 (2019-01-24): Merge pull request #665 from waveform-computing/pin-event-timing

Pin event timing (fixes #664)
closed 2018-09-18 by bennuttall