gpiozero/gpiozero #679: Add PumpkinPi board.

tuftii, 2018-10-08:
I'm new to developing for gpiozero and thought I'd add one of @modmypi's [Halloween boards](https://www.modmypi.com/raspberry-pi/led-displays-and-drivers-1034/led-boards-1040/halloween-pumpkin-solder-kit) as a starter.

I've got a handful of files at a fork [here](https://github.com/tuftii/python-gpiozero) if you'd like to see where I've got to.

Things I was unsure of:
- Testing, is there a document explaining how to write tests or is it just a matter of copying existing tests?
- Building docs, I've put some code in the docs for this board, but wasn't sure how to build it.
- If this is an appropriate thing to add into gpiozero.

Is there anything else I should do? Happy for comments and pointers :)

bennuttall, 2018-10-08:
This is awesome, thanks @tuftii.

> Testing, is there a document explaining how to write tests or is it just a matter of copying existing tests?

Not really - you don't need to copy the `LEDBoard` tests, but it might make sense to copy, say `SnowPi` or `StatusBoard` and test the general behaviour. Which pins should be high after `pumpk.left.on()` - that kind of thing. Be sure to test any edge cases.

> Building docs, I've put some code in the docs for this board, but wasn't sure how to build it.

The docs for docs are here: https://gpiozero.readthedocs.io/en/latest/development.html#building-the-docs

Does that make sense? They're pretty tied to Ubuntu, mind. Give me a shout if you have any issues.

> If this is an appropriate thing to add into gpiozero.

Yep :+1: 

> Is there anything else I should do? Happy for comments and pointers :)

Go ahead and open a PR. I'll review it (it looks fine) and merge once any minor issues resolved, and if the tests pass on travis (the PR will kick off a build).

bennuttall, 2018-10-08:
The only problem is I don't see us shipping v1.5 before halloween - so you might have to bundle a file with this class it in

tuftii, 2018-10-08:
Great, thanks Ben!

> They're pretty tied to Ubuntu, mind.

I'll have a go on my Ubuntu machine later, I'm on Windoze currently.

> I don't see us shipping v1.5 before halloween

No problem, I put a little class together outside of gpiozero for testing this morning, so I'll add that into the MMP repo temporarily, then I can remove it when 1.5 ships.

Tests - I'll add those in today/tomorrow and try to run them too.

closed 2019-02-01 by waveform80