gpiozero/gpiozero #681: Add JamHAT to gpiozero

tuftii, 2018-10-10:
After chatting with @bennuttall, I've opened up an issue to possibly add [JamHAT](https://www.modmypi.com/jam-hat) to GPIO Zero.

The board is 6 LEDs, a KC1206 buzzer and two buttons and looks like this:
[image]

Ben suggested any one of the nested options below to allow control over the lights together:

((red, amber, green), (red, amber, green), button, button, buzzer)
(((red, amber, green), (red, amber, green)), button, button, buzzer)
(((red, amber, green), (red, amber, green)), (button, button), buzzer)
((red, amber, green), (red, amber, green), (button, button), buzzer)
((red, red), (amber, amber), (green, green), (button, button), buzzer)
(red, amber, green, red, amber, green, button, button, buzzer)

(there are more)

The buzzer is 5V so I've been playing with it using the PWMOutputDevice class to try to get notes from it rather than a buzz, but the notes aren't that clean. Here's a short sample of what I'm doing, is there something different I could be doing in setup?

```
from gpiozero import PWMOutputDevice, Button
from time import sleep

pwm = PWMOutputDevice(20, frequency=2048)
button = Button(19, False)

try:
        while True:
                if button.is_pressed:
                        pwm.on()
                else:
                        pwm.off()
        sleep(0.1)
except:
        pwm.off()
        pwm.close()
        button.close()
```

@modmypi are producing the board with the York Pi Jam group and some basic examples using RPi.GPIO live [here](https://github.com/modmypi/Jam-HAT). It's available on the website, but we haven't done any publicity around it yet or polished the webpages, I was hoping to add it to GPIO Zero and create some more Jam friendly examples and worksheets to go along with it before we did anything more.

Over the next couple of days I'm going to fork the ModMyPi repo and change the examples over to GPIO Zero just to see what can be done. Any suggestions or pointers as to best approaches for the wider class would be ace :)

bennuttall, 2018-10-10:
It would be great to have a class for this, we just need to decide the format of the composite device.

I think if I were to choose now, I'd go for: ((red, amber, green), (red, amber, green), button, button, buzzer)

Therefore, the API would be:

```python
hat = JamHat()
hat.lights_1.on()
hat.lights_2.on()
hat.button_1.is_pressed
hat.button_2.is_pressed
hat.buzzer.on()
```

Ideally this type of buzzer would be its own generic class, and we can implement the tones in it. First suggested in #169.

tuftii, 2018-12-19:
I've finally got some time to have a look at implementing this class... 

So if we're going with (LEDBoard, LEDBoard, button, button, buzzer) is it best for me to extend the CompositeDevice class with JamHat or build a more generic class inbetween like TrafficLightsBuzzer then extend that with JamHat? I'm guessing the former, but thought I'd ask for an opinion before I run off and make a choice :)

lurch, 2018-12-21:
@tuftii I think `TrafficLightsBuzzer` was only implemented because there were several different add-on boards implementing the same features (just on different pins) so it made sense to add a shared parent class to avoid duplicating code. But since there aren't any other boards offering 6 LEDs, two buttons and a buzzer, I think there's probably no point in implementing a generic class :wink: 

@bennuttall Thinking back to the SnowPi, would having a single `lights` object be useful for turning on or off all the lights at once? And then if doing that, perhaps it'd make sense to group the buttons too? i.e. something like `(LEDBoard, ButtonBoard, Buzzer)`?
Also, unless there are specific silkscreen labels on the board, naming things `XXX_1` and `XXX_2` feels a bit arbitrary. @tuftii Are the buttons guaranteed to have fixed-colour caps? In which case perhaps the API could look like:
```python
hat = JamHat()
hat.lights.on()
hat.lights.top.red.blink()
hat.lights.bottom.off()
hat.lights.bottom.green.blink()
hat.buttons.red.is_pressed
hat.buttons.blue.is_pressed
hat.buzzer.on()
```

All just my suggestions of course, feel free to add as much salt as you want :)

bennuttall, 2019-01-08:
Well you have a choice between:

```
hat.lights_1.on()
hat.lights_1.red.blink()
```

and

```
hat.lights.on()
hat.lights.top.red.blink()
```

Do you want to group them by row (rgb / rgb) or by colour (rr / aa / gg). One of my options above covered this in another way:

> ((red, red), (amber, amber), (green, green), (button, button), buzzer)

could instead be:

(((red, red), (amber, amber), (green, green)), (button, button), buzzer)

tuftii, 2019-01-09:
Ok. @lurch sadly the caps on the buttons are removable, so they could be either way round. I had mine the opposite to the image above when I first got the board. It's easy enough to change them over as they're just clicky caps, but it could cause confusion if we code them as coloured buttons.

The LEDs are labelled on the silkscreen as LED1, LED3, LED5 on the top row and LED2, LED4, LED6 on the bottom row. We could go with this numbering, or the coloured light plus top or bottom.

I've built a simple class to get going outside of gpiozero and I went with the following:

```
class JamHat(CompositeOutputDevice):
    # Set up a star using GPIO Zero to build a class.
    # To use:
    def __init__(self, pwm=False, initial_value=False, pin_factory=None):
        super(JamHat, self).__init__(
            lights_1=LEDBoard(red=5, yellow=12, green=16,
                              pwm=pwm, initial_value=initial_value,
                              _order=('red', 'yellow', 'green'),
                              pin_factory=pin_factory),
            lights_2=LEDBoard(red=6, yellow=13, green=17,
                              pwm=pwm, initial_value=initial_value,
                              _order=('red', 'yellow', 'green'),
                              pin_factory=pin_factory),
            button_1=Button(19, pull_up=False, pin_factory=pin_factory),
            button_2=Button(18, pull_up=False, pin_factory=pin_factory),
            buzzer=Buzzer(20, pin_factory=pin_factory),
            pin_factory=pin_factory
            )
```
Meaning we can use:

```
jamhat.lights_1.on()
jamhat.lights_1.red.blink()
```
etc.

The buzzer I've implemented as a straight buzzer for now, but I'm having trouble getting it to work. It's a KC-1206 and it just seems to click when I do:

```
hat.buzzer.beep()
```
Is there something else I need to do, or is the test class I've made above a bit wrong?

bennuttall, 2019-01-10:
> The buzzer I've implemented as a straight buzzer for now, but I'm having trouble getting it to work. It's a KC-1206 and it just seems to click when I do `hat.buzzer.beep()`

Yeah it's not an on/off buzzer it's a tonal buzzer you need PWM for. You need to know what frequencies and duty cycle combinations make the tones it can play, and use `PWMOutputDevice` accordingly.

lurch, 2019-01-10:
...and be prepared for horrible noises if your pin-factory / pin means it uses software PWM instead of hardware PWM :wink: 

@bennuttall Perhaps it's worth adding a note to the current Buzzer docs explaining which types of buzzers it does and doesn't support? (I didn't even know there were different types!)

tuftii, 2019-01-10:
Ok, I've got it working using PWMOutputDevice. I'll play around with tones a bit and see what I can get out of it :)

bennuttall, 2019-01-12:
> @bennuttall Perhaps it's worth adding a note to the current Buzzer docs explaining which types of buzzers it does and doesn't support? (I didn't even know there were different types!)

Good point! Added in 2c9de273f61128c06ed1dcf45e64679970857fa6

tuftii, 2019-01-18:
Thanks for the quick chat today @bennuttall!

I've got the class working with the Jam Hat now and it plays the delightful rendition of Happy Birthday.

I've pasted the class and the demo script into a gist and it's [available here](https://gist.github.com/tuftii/879c9c62482b4b27b67f658da760c3ca).

The "magic number" we spoke about I haven't implemented and I've changed it to 0.5, but in it's current state it works :+1: .

I'll pop by the jam tomorrow and say hi a little after lunch if that's ok?

bennuttall, 2019-01-18:
Awesome thank you!

bennuttall, 2019-01-20:
Just wondering if we should provide [MIDI note numbers](http://www.inspiredacoustics.com/en/MIDI_note_numbers_and_center_frequencies) instead of (or as well as) notes, like Sonic Pi. e.g. MIDI 60 is middle C.

Could be as simple as adding another dict mapping MIDI numbers to notes and `play` would become:

```python
def play(self, note):
    if type(note) == int:
        note = self.midi_notes[note]
    self.frequency = self.notes[note]
    self.value = 0.5
```

tuftii, 2019-01-22:
We might be able to. I'm working on the PR today, so I'll have a go with MIDI tones whilst I'm here to see if we can.

bennuttall, 2019-01-22:
I've been thinking about this and I think we need to modify more when subclassing `PWMOutputDevice`. We shouldn't be setting value to 0.5 in `play`. Setting `value` should alter a device in an expected way.

Setting a PWMLED's value to 0.5 sets its duty cycle to half its frequency, which also sets its brightness to half, which is expected. Setting the value of a servo to 0 moves it to the centre position, uncoupled from its duty cycle. When `value` is set, we calculate what the new duty cycle should be and set it quietly in the background. Setting it to -1 turns it full left, 1 turns it full right, and you can set `source` to a sine wave to make it wave back-and-forth.

For PWMBuzzer, we should do something similar. I think it should be 0 to 1 not -1 to 1. Setting its value should probably set it to a particular note on a scale. That could be configurable, e.g:

```python
>>> bz = PWMBuzzer(20, min_note=48, max_note=72) # between C3 and C5
>>> bz.value = 0.5
>>> bz.frequency
261.63 (this is middle C)
>>> bz.frequency = 130.81
>>> bz.value
0
```

`play` would still set the note as it does now, but it would do so by setting the frequency. Also I think it could be possible to have `value` go to the nearest step, so you could do:

```python
for i in range(25): # 0 to 24, C3 to C5
    bz.value = i/16
```

(without rounding errors) which would (should) play one note at a time through a scale. My musical understanding is naive at best but I think something along these lines would be possible.

I think duty cycle would always be 0.5 when active (playing a note), 0 when not. We do something similar for servo - we have a `detatch` method to stop it jittering. But I think for this buzzer that would just be `bz.off()`.

I think I can implement this easy enough when I get time. Shouldn't hold anything up, should still be able to get it in the release.

lurch, 2019-01-22:
Perhaps you could also add a `note` property (with getters and setters) so that you could read/write `.value`, `.frequency` or `.note` (and possibly also `.midi`)  with everything getting calculated to/from `.value` ?

Although in the example above, having a `.value` of 0 corresponding to a frequency of "C3" rather than "off" feels a bit weird - maybe I just need to ponder it a bit more :wink:

bennuttall, 2019-01-22:
I know what you mean, but consider:

```python
>>> cpu = CPUTemperature(min_temp=50)
>>> cpu.value
0.0
>>> cpu.temperature
50
```

Also `.value = 0` doesn't turn a Servo off, it moves it to a particular position.

bennuttall, 2019-01-23:
Eugh, just thinking you would need to be able to turn the buzzer off by setting `value` otherwise `source` could never play a tune with pauses. Maybe `None`...

lurch, 2019-01-23:
Not sure if I'm keen on a `value` of 0..1 _and_ None. What if it was a tuple of (0..1, bool) indicating `(value, enabled)` ? Could that be made to work with sourcetools?

bennuttall, 2019-01-23:
@waveform80 agrees and points out that's what we do now for Servo! (I hadn't realised this)

lurch, 2019-01-23:
I guess that means that a Servo could also 'play' the same tune as a buzzer :wink:

tuftii, 2019-01-23:
Right, I've uploaded what I've got so far, as I've got a bit stuck. Fully expecting it to fail tests, but I'm not sure what's wrong :sad:

bennuttall, 2019-01-23:
It's somehow removed all the history for those files. I'm not sure how. Might be worth trying again. How are you doing it? git command line on linux, github for windows app, something else?

bennuttall, 2019-02-01:
Merged! 9df0c621630fc2b4df0ab5f955d9c93690232fec

closed 2019-02-01 by bennuttall

tuftii, 2019-02-04:
Yaay! :tada:

bennuttall, 2019-02-04:
Release imminent! @waveform80 is working on improving the tones interface.

tuftii, 2019-02-04:
Oooooooo! Exciting.

I have one last question, did we need an hat.off() which calls hat.buzzer.play(None) as right now hat.off() doesn't stop the buzzer?

bennuttall, 2019-02-04:
Hmm, we decided `hat.on()` should't turn the buzzer on (because what note would it play?) but perhaps it should turn the buzzer off? @waveform80 any thoughts?

bennuttall, 2019-02-04:
Perhaps `buzzer.play()` with no args should play the mid note? Then maybe `hat.on()` would call `hat.buzzer.play()` with no args :man_shrugging:

lurch, 2019-02-10:
I don't have any preference as to whether `hat.on()` should turn the buzzer on too (I've not been playing with buzzers, so I dunno how annoying they are :wink: ) but IMHO `hat.off()` should definitely turn the buzzer off. I guess this is one of those pragmatism vs. purism decisions :)