gpiozero/gpiozero #685: prevent when_pressed event when_held is used for the same button too

marcohorstmann, 2018-10-25:
Operating system: e.g. Raspbian Stretch
Pi model: e.g. Pi 2 Model B+
GPIO Zero version: 1.4.1
Pin factory used: <gpiozero.pins.rpigpio.RPiGPIOFactory object at 0x7607e510>

Hi, I tried to implement a media player control via GPIO buttons. I want to
push a button once and just a push and he shut e.g. switch to the next song with Button called next,
but on the other hand if I want to seek in this song I want to hold down the next button for some time.
In the moment it happens that when_pressed event is fired up he goes to the next song and then seems because when_held event is triggered. It would be cool if it is possible to wait e.g. the hold_time to fire up the when_pressed event. Any suggestions how I can do it?

```
#!/usr/bin/python3
from gpiozero import Button
from signal import pause
from subprocess import check_call

# 2017-12-12
# This script was copied from the following RPi forum post:
# https://forum-raspberrypi.de/forum/thread/13144-projekt-jukebox4kids-jukebox-fuer-kinder/?postID=312257#post312257
# 2018-10-25 modified to implement functions for a pressed down button
#

def def_shutdown():
    check_call("./scripts/playout_controls.sh -c=shutdown", shell=True)

def def_volU():
    check_call("./scripts/playout_controls.sh -c=volumeup", shell=True)

def def_volD():
    check_call("./scripts/playout_controls.sh -c=volumedown", shell=True)

def def_vol0():
    check_call("./scripts/playout_controls.sh -c=mute", shell=True)

def def_next():
    check_call("./scripts/playout_controls.sh -c=playernext", shell=True)

def def_prev():
    check_call("./scripts/playout_controls.sh -c=playerprev", shell=True)

def def_halt():
    check_call("./scripts/playout_controls.sh -c=playerpause", shell=True)

def def_seekforward():
    check_call("./scripts/playout_controls.sh -c=playerseek -v=+10", shell=True)

def def_seekbackward():
    check_call("./scripts/playout_controls.sh -c=playerseek -v=-10", shell=True)

shut = Button(3, hold_time=2)
vol0 = Button(13,pull_up=True)
volU = Button(16,pull_up=True,hold_time=0.3,hold_repeat=True)
volD = Button(19,pull_up=True,hold_time=0.3,hold_repeat=True)
next = Button(26,pull_up=True,hold_time=0.5,hold_repeat=True)
prev = Button(20,pull_up=True,hold_time=0.5,hold_repeat=True)
halt = Button(21,pull_up=True)

#If shutdown button was held for hold_time of 2 secounds shutdown will be executed
shut.when_held = def_shutdown
#If mute button was pressed the system will be muted
vol0.when_pressed = def_vol0
#When the Volume Up button was pressed it will raise the volume
volU.when_pressed = def_volU
#When the Volume Up button was held for more than 0.3 seconds every 0.3 seconds he will call a raise of volume
volU.when_held = def_volU
#When the Volume Down button was pressed it will lower the volume
volD.when_pressed = def_volD
#When the Volume Down button was held for more than 0.3 seconds every 0.3 seconds he will lower the volume
volD.when_held = def_volD
#When the Next button will be pressed next item will be played
next.when_pressed = def_next
next.when_held = def_seekforward
prev.when_pressed = def_prev
prev.when_held = def_seekbackward
halt.when_pressed = def_halt

pause()
```

bennuttall, 2019-01-12:
Sorry for the delay, I remember seeing this and remembering I'd answered this question somewhere before but I couldn't find it and I never got around to writing out a solution.

Anyway, the way to achieve this is to *not* set a callback on `when_pressed`, and instead use `when_released` to work out whether it had been held or just pressed:

```python
from gpiozero import Button

was_held = False

def held():
    global was_held
    was_held = True
    print("button was held not just pressed")

def released():
    global was_held
    if not was_held:
        pressed()
    was_held = False

def pressed():
    print("button was pressed not held")

btn = Button(2)

btn.when_held = held
btn.when_released = released
```

Alternatively, without using globals (unfortunately this requires subclassing to allow a new attribute to be written to, because gpiozero prevents this usually):

```python
from gpiozero import Button

class MyButton(Button):
    def __init__(self, *args, **kwargs):
        self.was_held = False
        super(MyButton, self).__init__(*args, **kwargs)

def held(btn):
    btn.was_held = True
    print("button was held not just pressed")

def released(btn):
    if not btn.was_held:
        pressed()
    btn.was_held = False

def pressed():
    print("button was pressed not held")

btn = MyButton(2)

btn.when_held = held
btn.when_released = released
```

Either way, here's an example showing both pressed and hold actions, using MockPin to simulate button state changes:

```python
In [1]: %run test.py

In [2]: btn.when_held = held 

In[3]: btn.when_released = released                                                                 

In [4]: btn.pin.drive_low()                                                                          

button was held not just pressed
In [5]: btn.pin.drive_high()                                                                         

In [6]: btn.pin.drive_low(); btn.pin.drive_high()                                                    
button was pressed not held
```

bennuttall, 2019-01-14:
Closing this as I think it's covered, but feel free to comment if you have any more questions

closed 2019-01-14 by bennuttall

bennuttall, 2019-01-14:
Actually here's a better OO solution that doesn't use globals and doesn't require subclassing:

```python
from gpiozero import Button

Button.was_held = False

def held(btn):
    btn.was_held = True
    print("button was held not just pressed")

def released(btn):
    if not btn.was_held:
        pressed()
    btn.was_held = False

def pressed():
    print("button was pressed not held")

btn = Button(2)

btn.when_held = held
btn.when_released = released
```

marcohorstmann, 2019-01-15:
Thx for this Code example. I will try to implement this change to the project i working on when I have more time. In the Moment i‘m a little bit busy.