gpiozero/gpiozero #686: DistanceSensor does not work reliably (or at all)  on Pi Zero

topshed, 2018-10-26:
(Possibly a dupe of #644 but this is over all distances)

Operating system: Latest Raspbian Stretch Lite
Pi model: Pi Zero 1.3
GPIO Zero version: e.g. 1.4.1
Pin factory used: e.g. RPiGPIO

```from gpiozero import DistanceSensor
from time import sleep

ds = DistanceSensor(15,14)

while True:
    print(ds.distance)
    sleep(0.1)
```
The results when starting with no obstacle then placing one at about 5 cm for sensor and moving back and forwards are:

1.0
1.0
1.0
1.0
1.0
1.0
1.0
1.0
1.0
1.0
1.0
1.0
1.0
1.0
1.0
1.0
1.0
1.0
1.0
1.0
1.0
1.0
1.0
0.5
0.0
0.0
0.0
0.0
0.0
0.0
0.0
0.0
0.0
0.0
0.0
0.0
0.0
0.0
0.0
            
In contrast, using the same set-up with the hcsr04sensor library gives good results (but note that I'm  using 3 samples to give faster results  and thus sacrificing accuracy (default is 11)). 

```python
from hcsr04sensor import sensor
from time import sleep

d = sensor.Measurement(14,15)
while True:
    print(d.raw_distance(sample_size=3))
    sleep(0.1)
```
244.04195069250022
242.5322103902967
242.654953504297
242.58130763589685
242.56494188736346
243.25230332576507
5.60936030981303
8.285160195019246
17.126755840173118
5.9694067775472
7.446415582683964
8.526554985886474
8.83750420802053
10.28178151609055
11.034605948625632
11.910173495161

bennuttall, 2019-01-15:
Let's test this with the pin timing branch, which is planned for v1.5. It should be fixed there.

https://github.com/waveform-computing/gpio-zero/tree/pin-event-timing

bennuttall, 2019-01-25:
This *should* be fixed on master now. Two big improvements to the way we calculate distance thanks to great effort from @waveform80. Please test if you can:

```
git clone https://github.com/RPi-Distro/python-gpiozero
cd python-gpiozero
sudo pip3 install .
```

The v1.5.0 release is expected very soon.

closed 2019-02-01 by bennuttall

hyansuper, 2019-02-16:
I notice this issue is closed, does that mean it's fixed?
I've upgraded to v1.5 but the issue remains

bennuttall, 2019-02-16:
What Pi model are you on? Zero?

Which Python version?

Can you confirm the GPIO Zero version you're using is v1.5 by running [this](https://gpiozero.readthedocs.io/en/stable/faq.html#how-can-i-tell-what-version-of-gpiozero-i-have-installed) in the same environment you're running your code (to make sure it's not e.g. installed for Python 2 when you're using 3)

Also we now recommend using the pigpio pin factory, especially on Pi 1/Zero as noted here: https://gpiozero.readthedocs.io/en/stable/api_input.html#distancesensor-hc-sr04

```bash
$ sudo pigpiod
$ export GPIOZERO_PIN_FACTORY=pigpio
$ python3 distance.py
```

See if you get better results and let us know.

hyansuper, 2019-02-17:
I am on Pi Zero, and v1.5 is conformed:
`>>> from pkg_resources import require
>>> require('gpiozero')
[gpiozero 1.5.0 (/usr/local/lib/python3.5/dist-packages), colorzero 1.1 (/usr/local/lib/python3.5/dist-packages)]
>>> require('gpiozero')[0].version
'1.5.0'`
and I have export GPIOZERO_PIN_FACTORY=pigpio.
The reading is between 50~60 regardless of the actual distance.
But if I enable remote GPIO and run the code on windows(also python3 and gpiozero v1.5), the reading is constantly 100.

francescoatmitie, 2019-05-26:
Similar issue but with RaspPI 3 B+, see #770