gpiozero/gpiozero #687: 'RuntimeError: Failed to add edge detection' when running script for 2nd time

glynhudson, 2018-10-29:
Operating system: e.g. Raspbian Stretch
Pi model: Pi 3 Model B+
GPIO Zero version: 1.4.1
Pin factory used: e.g. RPiGPIO

I'm testing a simple GPIO push button script. After a fresh reboot the first time the script is run it works great. However, after stopping the script than trying to run it again I get error `RuntimeError: Failed to add edge detection`. Do I need to remove the edge detection interrupt? I though gpiozero should do this automatically?  

```
from gpiozero import Button
from signal import pause

print ("Push button test")

def buttonPressLong():
    print ("long press")

def buttonPress():
   print "short press"

push_btn = Button(23, pull_up=False, hold_time=5, bounce_time=0.1)
push_btn.when_pressed = buttonPress
push_btn.when_held = buttonPressLong
   
pause()
```

```
Traceback (most recent call last):
  File "push_and_hold.py", line 16, in <module>
    push_btn = Button(23, pull_up=False, hold_time=5, bounce_time=0.1)
  File "/usr/lib/python2.7/dist-packages/gpiozero/devices.py", line 95, in __call__
    self = super(GPIOMeta, cls).__call__(*args, **kwargs)
  File "/usr/lib/python2.7/dist-packages/gpiozero/input_devices.py", line 303, in __init__
    pin, pull_up, bounce_time, pin_factory=pin_factory
  File "/usr/lib/python2.7/dist-packages/gpiozero/mixins.py", line 338, in __init__
    super(HoldMixin, self).__init__(*args, **kwargs)
  File "/usr/lib/python2.7/dist-packages/gpiozero/input_devices.py", line 100, in __init__
    self.pin.when_changed = self._fire_events
  File "/usr/lib/python2.7/dist-packages/gpiozero/pins/__init__.py", line 388, in <lambda>
    lambda self, value: self._set_when_changed(value),
  File "/usr/lib/python2.7/dist-packages/gpiozero/pins/pi.py", line 289, in _set_when_changed
    self._enable_event_detect()
  File "/usr/lib/python2.7/dist-packages/gpiozero/pins/rpigpio.py", line 219, in _enable_event_detect
    bouncetime=self._bounce)
RuntimeError: Failed to add edge detection
```

I'm running the latest RPi.GPIO

```
sudo apt-get install python-rpi.gpio python3-rpi.gpio
Reading package lists... Done
Building dependency tree       
Reading state information... Done
python-rpi.gpio is already the newest version (0.6.4~stretch-1).
python3-rpi.gpio is already the newest version (0.6.4~stretch-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```

bdilloughery, 2018-11-03:
I have not yet tried as much as you, but I am having the same or a similar issue.

I don't have my Pi handy, as it is in my classroom ... but 33 new raspberry pi's with the current Raspian on them, all get the same errror when I try and run a simple 4 line python program such as:

`
from gpiozero import Button

button = Button(2)

button.wait_for_press()    
print("Button was pressed")`

bcroston, 2018-11-03:
Are you running your program from the command line or from a GUI (like Idle / Mu / Thonny)?

bdilloughery, 2018-11-03:
Running then all in Python 3 IDLE

bdilloughery, 2018-11-03:
... and mine doesn't have a loop. It doesn't even run once

bcroston, 2018-11-03:
I've fixed it in the latest development RPi.GPIO.  To try it (in Raspbian squeeze):

$ sudo apt-get install python-dev python3-dev
$ sudo apt-get install mercurial
$ sudo apt-get install python-pip python3-pip
$ sudo apt-get remove python-rpi.gpio python3-rpi.gpio
$ sudo pip install hg+http://hg.code.sf.net/p/raspberry-gpio-python/code#egg=RPi.GPIO
$ sudo pip3 install hg+http://hg.code.sf.net/p/raspberry-gpio-python/code#egg=RPi.GPIO

To revert back to the default version in Raspbian:

$ sudo pip uninstall RPi.GPIO
$ sudo pip3 uninstall RPi.GPIO
$ sudo apt-get install python-rpi.gpio python3-rpi.gpio

glynhudson, 2018-11-04:
Hi @bcroston,

Thanks, for reply. I've just tried to test. I executed the commads above (omitting the python3 since I'm using 2.7). I managed to succesfully uninstall python-rpi-gpio. However, when I tired to install your dev verson I get the message:

```
$ sudo pip install hg+http://hg.code.sf.net/p/raspberry-gpio-python/code#egg=RPi.GPIO
Requirement already satisfied: RPi.GPIO from hg+http://hg.code.sf.net/p/raspberry-gpio-python/code#egg=RPi.GPIO in /usr/local/lib/python2.7/dist-packages
```

bennuttall, 2018-11-04:
It must still be installed somehow. Perhaps you installed with pip previously? You can try to remove it (`sudo apt purge python-rpi.gpio` and `sudo pip uninstall rpi.gpio`) or just force an upgrade: `sudo pip install hg+http://hg.code.sf.net/p/raspberry-gpio-python/code#egg=RPi.GPIO --upgrade`

glynhudson, 2018-11-05:
Fantastic, I can confirm after upgrading to the dev version of RPi.GPIO 0.6.4 this issue has been fixed:

`sudo pip install hg+http://hg.code.sf.net/p/raspberry-gpio-python/code#egg=RPi.GPIO --upgrade`

I can run my example code above to attach an interrupt. Nice work 👍 

Will this fixe be merged into the 0.6.4 or 0.6.5 offical release? Would you recomend I continue running 0.6.3 stable until this fix is depolyed? 

Why is there both an apt version of rpi-gpio and a pip version? I'm assuming the pip version is easier for you to update since this does not need to go via the package manager.

bdilloughery, 2018-11-05:
This fixed all my issues. Thank you bcroston!

> I've fixed it in the latest development RPi.GPIO. To try it (in Raspbian squeeze):
>
> $ sudo apt-get install python-dev python3-dev
> $ sudo apt-get install mercurial
> $ sudo apt-get install python-pip [...]

closed 2018-11-07 by bennuttall

dlivingstone, 2019-01-06:
As a note... I had the same issue with one machine only (my own) out of my classroom - which was odd, as they all used the same image initially - so not sure how this happened. The problem affected all gpiozero inputs tried (e.g. distance sensor, button, temperature)

The fixes described above didnt seem to work for me, but a more recent update to latest version of everything has now worked.

There is a workaround that can be employed if you need to know your scripts will re-run reliably on machines that might not have been updated - it requires a bit of editing to the code though.

If you re-write the code so that the button's close method is called,e.g. push_btn.close(), before the end of the program you should be able to re-run. If ctrl-c is required to exit a loop, then you'll also need to add a try/except block of code, and put close in the except block.

dony71, 2020-06-11:
why this issue closed? what the solution on this?

bdilloughery, 2020-06-11:
> why this issue closed? what the solution on this?

It was just an error in that current build version of either Raspian or GPIOZERO module. I cant remember. I think gpiozero is created and maintained by the single developer, mb bcroston above. There were temp fixes, but the next version fixed the error. As long as you are running updated Raspian and gpiozero you shouldn't get this error anymore ... or at least not for the same reason.

bennuttall, 2020-06-11:
The above is correct, if you replace "gpiozero" with "RPi.GPIO".