gpiozero/gpiozero #688: Support device detection under aarch64

Martchus, 2018-11-02:
Under aarch64, `/proc/cpuinfo` does not contain the revision number leading to the error ["unable to locate Pi revision in /proc/cpuinfo"](https://github.com/RPi-Distro/python-gpiozero/blob/e8bf5b500f9f2ea55d06869a86749b9c91691f68/gpiozero/pins/local.py#L67).

This [is not likely to change](https://github.com/raspberrypi/linux/issues/2110#issuecomment-334459990). So I suppose it would be the best to add a fallback to `/proc/device-tree`.

I tested the following C implementation and it seems to work: https://github.com/jgarff/rpi_ws281x/pull/316/files#diff-6ee9c21b70f61e8effb715dd4df0ab53R330

So to solve this issue, one would simply need to port it to Python and put as fallback before raising the exception.

I'm wondering why this hasn't been implemented yet as it seems to be an easy task. But if you agree to have device detection work under aarch64 in the way I suggested, I would go ahead and create PR.

Martchus, 2018-11-02:
I am using Arch Linux ARM (aarch64 version) and built the package for the Python library provided by this repository myself. I've been using the following build script: https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=python-gpiozero

Actually I've used a slightly adjusted version to get the latest commit available in this repository (in the hope the support has already been added since the last release).

bennuttall, 2018-11-02:
Great. No issue with getting this fixed - but I'm a little busy right now, so if you want to try to patch it yourself, feel free to send a PR.

bombadil, 2018-12-29:
Ping. Any update?

Martchus, 2018-12-29:
Not from my side. So the tests are still missing but one can locally apply the first monitor from my PR in the mean time.

lurch, 2019-01-21:
Note that RPi.GPIO and pigpio (two of the pin libraries supported by GpioZero) don't appear to support aarch64 either yet...
https://sourceforge.net/p/raspberry-gpio-python/tickets/161/
https://github.com/joan2937/pigpio/pull/255

closed 2019-01-23 by bennuttall

lurch, 2019-01-23:
Just for reference: Fixed by #689

bcroston, 2019-03-16:
I've fixed RPi.GPIO for aarch64 now (please test and confirm).
You will need to install the development copy using commands along the lines of:
$ sudo apt install mercurial
$ pip3 install --upgrade hg+http://hg.code.sf.net/p/raspberry-gpio-python/code#egg=RPi.GPIO