gpiozero/gpiozero #690: .wait_for_press() not working as expected

qqgg231, 2018-11-07:
Operating system: Raspbian Stretch
Pi model:  Pi 3 Model B
GPIO Zero version: 1.4.1
Pin factory used: GPIOZERO_PIN_FACTORY=pigpio
Using remotegpio from vs_code (Windows)

My problem is that `print("Button green(25) Pressed")` is never printed.

Earlier when I used `GPIO.wait_for_edge()`  it was working ok, so I think it is bug in gpiozero

```python
import time
import datetime
from gpiozero import LED, Button
from time import sleep

def main():

    global i
    global stop

    bRed = Button(24)
    bGreen = Button(25)

    def bRedPressed():
        print("Button red(24) pressed!")
        global stop
        global i
        i = 0
        stop = True
        bGreen.wait_for_press()  # waits for click, but stops here, although I click many times
        print("Button green(25) Pressed")  # never printed !
        i = 60*60
        stop = False
    bRed.when_pressed = bRedPressed

#script starts here
    bGreen.wait_for_press() # waits for click,  button is working, when I click it goes to next line

    i = 60*60
    stop = False
    while True:
        time.sleep(1)
        if not stop:
            i += 1
        print(str(i))

if __name__ == '__main__':
    main()
```

bennuttall, 2018-11-07:
Generally speaking, if someone reports that a single commonly used function does not work, I expect the report to be as minimal as possible. If you found that `wait_for_press` doesn't work I want to see the smallest possible example that shows it does not work as expected, e.g:

```python
from gpiozero import Button
btn = Button(2)
print("waiting for press")
btn.wait_for_press()
print("pressed")
```

I would expect this to print "waiting for press", then print "pressed" once I pressed the button. If that works, but your code doesn't, then it's not `wait_for_press` that doesn't work, it's something in your code.

I would try to reduce your problem down as much as possible. Start from my simple example, and work towards what you want, feature by feature. When you get to a point where it stops doing what you expected, then investigate this change.

It's hard to tell what you're trying to do - you're mixing paradigms. Please be aware that callbacks are threaded and dealing with this kind of logic inside threaded code can lead to unexpected behaviour.

qqgg231, 2018-11-08:
Thank you for very informative answers! I will check code and will let you know :-)