gpiozero/gpiozero #695: Performance for blinking once compared to RPi.GPIO while implementing HX711 device

vtalpaert, 2018-11-19:
Hello, I am writing a HX711 weight scale class using `gpiozero.SmoothedInputDevice`.

The HX711 is a 24-Bit Analog-to-Digital Converter (ADC) for Weigh Scales [(datasheet)](https://cdn.sparkfun.com/datasheets/Sensors/ForceFlex/hx711_english.pdf):
 - It uses a variant of the SPI interface since the timing of the clock can set the converter's gain.
 - The device goes to sleep if the clock is high for longer than 60 us.

For reference, you can find [here](https://github.com/gandalf15/HX711/blob/dev/HX711_Python3/hx711.py) a working implementation using RPi.GPIO. It manually controls the pins. [Some Arduino C++ implementation](https://github.com/dudapickler/hx711_SPI) proposed to connect the device's clock to the SPI MOSI to send specific bytes.

In my code, I need to blink the clock once before reading the data out, but the blink must last less than 60 us. My issue is that `gpiozero` is too slow. Is here a snippet comparing the RPi.GPIO to gpiozero:

Operating system: Raspbian
Pi model: Pi Zero W
GPIO Zero version: 1.4.1
Pin factory used: RPiGPIO

```python
import time
import numpy as np

from gpiozero import DigitalOutputDevice
import RPi.GPIO as GPIO

objective = 0.00006  # 60 us

def print_info(name, times):
    failed = len(list(filter(lambda t: t > objective, times)))
    print(name, '\t', 'mean: {:.8f}'.format(np.mean(times)),
          '\t', 'std: {:.8f}'.format(np.std(times)),
          '\t', 'failed:', failed
    )

def performance_zero(n, pin_number):
    times = []
    pin = DigitalOutputDevice(pin=pin_number)
    for _ in range(n):
        start_counter = time.perf_counter()
        pin.on()
        pin.off()
        end_counter = time.perf_counter()
        times.append(end_counter - start_counter)
    print_info('zero', times)

def performance_write(n, pin_number):
    times = []
    pin = DigitalOutputDevice(pin=pin_number)
    for _ in range(n):
        start_counter = time.perf_counter()
        pin._write(True)
        pin._write(False)
        end_counter = time.perf_counter()
        times.append(end_counter - start_counter)
    print_info('write', times)

def performance_blink(n, pin_number):
    times = []
    pin = DigitalOutputDevice(pin=pin_number)
    for _ in range(n):
        start_counter = time.perf_counter()
        pin.blink(on_time=0, off_time=0, n=1, background=False)
        end_counter = time.perf_counter()
        times.append(end_counter - start_counter)
    print_info('blink', times)

def performance(n, pin_number):
    times = []
    GPIO.setup(pin_number, GPIO.OUT)
    for _ in range(n):
        start_counter = time.perf_counter()
        GPIO.output(pin_number, True)
        GPIO.output(pin_number, False)
        end_counter = time.perf_counter()
        times.append(end_counter - start_counter)
    print_info('gpio', times)

if __name__ == '__main__':
    pin_number = 16
    n = 100
    performance_zero(n, pin_number)
    performance_write(n, pin_number)
    performance_blink(n, pin_number)
    performance(n, pin_number)

```

```bash
$ python3 benchmark.py 
zero 	 mean: 0.00058894 	 std: 0.00019848 	 failed: 100
write 	 mean: 0.00021100 	 std: 0.00008171 	 failed: 100
blink 	 mean: 0.00444331 	 std: 0.00065813 	 failed: 100
gpio 	 mean: 0.00004225 	 std: 0.00013700 	 failed: 2
```

Am I using the library wrong ? I know RPi.GPIO [warns not to use the lib for time critical applications](https://pypi.org/project/RPi.GPIO/), but still getting this to work would be nice. Lots of people are struggling with this device.

bennuttall, 2018-11-19:
Critical speed isn't one of the main objectives of the library, but let's see what's going on.

I see you're using `DigitalOutputDevice`. This introduces the `blink` method which has the potential to be threaded. You can see from the [class implementation](https://github.com/RPi-Distro/python-gpiozero/blob/master/gpiozero/output_devices.py#L129) that this complicates things, and adds many more instructions to be followed (i.e. before every on/off call, check if there's a blink thread and kill it). You can avoid this by using the more abstract [`OutputDevice`](https://github.com/RPi-Distro/python-gpiozero/blob/master/gpiozero/output_devices.py#L19), and some quick testing tells me you pretty much get RPi.GPIO speeds:

```python
from gpiozero import OutputDevice  # changed this import
import RPi.GPIO as GPIO

class DigitalOutputDevice(OutputDevice):
    def blink(self, on_time=1, off_time=1, n=1, background=None):
        # a non-threaded blink method
        for _ in range(n):
            self.on()
            time.sleep(on_time)
            self.off()
            time.sleep(off_time)
```

My results:

```
pi@raspberrypi:~ $ python3 timing.py 
zero 	 mean: 0.00003223 	 std: 0.00000443 	 failed: 1
write 	 mean: 0.00002855 	 std: 0.00000117 	 failed: 0
blink 	 mean: 0.00004630 	 std: 0.00000563 	 failed: 2
gpio 	 mean: 0.00000458 	 std: 0.00000044 	 failed: 0
pi@raspberrypi:~ $ python3 timing.py 
zero 	 mean: 0.00003182 	 std: 0.00000198 	 failed: 0
write 	 mean: 0.00002874 	 std: 0.00000174 	 failed: 0
blink 	 mean: 0.00004645 	 std: 0.00000415 	 failed: 1
gpio 	 mean: 0.00000477 	 std: 0.00000041 	 failed: 0
pi@raspberrypi:~ $ python3 timing.py 
zero 	 mean: 0.00003091 	 std: 0.00000201 	 failed: 0
write 	 mean: 0.00002788 	 std: 0.00000133 	 failed: 0
blink 	 mean: 0.00004556 	 std: 0.00000458 	 failed: 1
gpio 	 mean: 0.00000458 	 std: 0.00000043 	 failed: 0
```

bennuttall, 2019-01-12:
Closing as I think this has been answered, but feel free to comment if you have further questions.

closed 2019-01-12 by bennuttall

vtalpaert, 2019-01-14:
Hi Ben, thanks for the response.
Indeed this is more worthy of a `question` tag.

The blink of your proposed `DigitialOutputDevice` class is essentially doing the same as the "zero" function. Although I guess calling twice `time.sleep(0)` is adding 10us.

Nonetheless, the issue is coming from my Raspberry Zero. In execution time, compared to your Pi my RPi.GPIO is 10x slower, when gpiozero is 100x slower ! For my time critical application, gpiozero is not suited on Raspberry Zero.

Still the `SmoothedInputDevice` use of a background thread is great, especially when several readings are needed. I tried a [mixed solution](https://github.com/vtalpaert/autobar/blob/master/hardware/hx711.py) of a RPi.GPIO code with a gpiozero background thread that works great.