gpiozero/gpiozero #697: when_held callback repeatedly called even after button has been released

rommar, 2018-11-25:
Hello. I'm trying to use when_pressed and when_held callbacks simultaneously. Sometimes when_held callback is called even after I release the button. This behavior isn't stable but occures approximately half the cases after I press and release the button. The button connects 3.3 V to a GPIO pin when pressed, so pull-down configuration is used.

```python
import gpiozero
import time

def pressed(b):
    print 'Pressed', b

def held(b):
    print 'Held', b, b.pressed_time

button = gpiozero.Button(22, pull_up=False, bounce_time=0.050, hold_time=1, hold_repeat=True)
button.when_pressed = pressed
button.when_held = held

while(True):
    time.sleep(2)
    print 'Waiting...'
```

Below is example output.
I marked the moment when I released the button last time. Note that is_active=False after that. Does it mean the button isn't pressed? Why the callback is called then?
```
$ python ./buttons.py
Waiting...
Waiting...
Pressed <gpiozero.Button object on pin GPIO22, pull_up=False, is_active=True>
Waiting...
Pressed <gpiozero.Button object on pin GPIO22, pull_up=False, is_active=True>
Waiting...         <<---- I released the button at this moment
Held <gpiozero.Button object on pin GPIO22, pull_up=False, is_active=False> 1.02130889893
Held <gpiozero.Button object on pin GPIO22, pull_up=False, is_active=False> 2.0249080658
Waiting...
Held <gpiozero.Button object on pin GPIO22, pull_up=False, is_active=False> 3.0283370018
Held <gpiozero.Button object on pin GPIO22, pull_up=False, is_active=False> 4.03196501732
Waiting...
Held <gpiozero.Button object on pin GPIO22, pull_up=False, is_active=False> 5.03540992737
Held <gpiozero.Button object on pin GPIO22, pull_up=False, is_active=False> 6.03898000717
Waiting...
Held <gpiozero.Button object on pin GPIO22, pull_up=False, is_active=False> 7.04250502586
Held <gpiozero.Button object on pin GPIO22, pull_up=False, is_active=False> 8.04601311684
Waiting...
```

Versions and other technical information:
```
$ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

$ cat /proc/device-tree/model
Raspberry Pi Model B Rev 2

$ python -i
Python 2.7.13 (default, Sep 26 2018, 18:42:22)
[GCC 6.3.0 20170516] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> from pkg_resources import require
>>> require('gpiozero')[0].version
'1.4.1'
>>> from gpiozero import Device
>>> Device.pin_factory
<gpiozero.pins.rpigpio.RPiGPIOFactory object at 0xb63fc710>

```

Am I doing something wrong here or this is a bug?

lurch, 2019-02-11:
Sounds like this might be similar to #550 ?

waveform80, 2019-02-13:
Yes ... sounds very similar. This may well be RPi.GPIO dropping edges. I'd be very interested to see if the same thing happens with the updated "native" pin driver in 1.5.0 or with the "pigpio" driver. RPi.GPIO uses a single thread to both listen for epoll events (indicating a pin edge) *and* for dispatching callbacks. pigpio and the new "native" driver both use one thread (/process) for listening for edges, and a separate thread (/process) for dispatch so theoretically they don't have the issue of a callback suppressing edges (although the "native" driver might still be slow enough to lose them, being pure Python).

lurch, 2019-02-13:
@waveform80 Would the when_held handler checking that the button is still pressed (to workaround issues like the one presented here) add too much overhead?

bennuttall, 2019-02-13:
Surely when the `when_held` callback is called the button will *always* still be pressed? Unless the user only holds the button down for say *exactly* one second and releases the very moment the callback is fired?

lurch, 2019-02-13:
> Surely when the `when_held` callback is called the button will _always_ still be pressed?

See the log above, where it prints out `Held ... is_active=False`.
I've not looked at the code in detail, but my (possibly incorrect) assumption was that the falling-edge is seen and triggers the 'press' event, which in turn triggers the "held" state after hold_timeout seconds, but because (for whatever reason) the rising-edge is missed, the 'release' event doesn't trigger, and so the "held" state persists; even though reading the button directly reveals that the button is no longer being pressed.

Obviously in an ideal world each falling-edge event would be matched by a corresponding rising-edge event, but due to the non-realtime nature of Linux, the interpreted nature of Python, etc. it seems that edges might be being missed / suppressed as @waveform80 suggests above.

waveform80, 2019-02-21:
> I've not looked at the code in detail, but my (possibly incorrect) assumption was that the falling-edge is seen and triggers the 'press' event, which in turn triggers the "held" state after [...]

@lurch's assessment is absolutely correct - this is exactly what happens. With RPi.GPIO, this is particularly likely if a callback takes a long time (i.e. several milliseconds) to execute because epoll in edge-trigger mode "drops" edges if nothing is actively waiting for them and RPi.GPIO uses a single thread for monitor and dispatch. In other words, it won't be actively waiting for an edge until the last callback returns (incidentally, this doesn't include `when_held` callbacks which are dispatched from another thread started by the initial pressed callback).

With the new (1.5.0 onwards) "native" pin driver this is *possibly* less likely as I used separate monitor and dispatch threads in its implementation ... but it's all still pure Python, and Python threading primitives are very slow so it's possible that could still lose edges. Under the "pigpio" driver I'd consider it *very* unlikely (maybe impossible?) as it uses DMA to regularly sample the GPIO registers for state changes.

I'm in two minds as to what to do about this. With RPi.GPIO, we could trivially work around it by testing the pin status before each `when_held` dispatch. It's cheap, and it's works around the problem. But with the "pigpio" driver that's actually very expensive (potentially involves a remote network round-trip) and is also completely unnecessary with that driver.

@rommar can you test whether this is still happening under 1.5.0? There were some event handling changes there that *might* mitigate this under the default RPi.GPIO driver (specifically `EventMixin._fire_events` got changed quite a bit). This release is now in Raspbian so a simple "apt update", "apt upgrade" should be sufficient for you to test it. If it's still happening regularly, I'll need to think about how we can fix this without costing too much under "pigpio"...

lurch, 2019-02-22:
> @lurch's assessment is absolutely correct - this is exactly what happens.

Woohoo!

> I'm in two minds as to what to do about this. With RPi.GPIO, we could trivially work around it by testing the pin status before each `when_held` dispatch. It's cheap, and it's works around the [...]

Perhaps each PinFactory could have a `might_miss_edges` property? :slightly_smiling_face:

matthijskooijman, 2023-03-22:
I'm also seeing this issue (or at least the same symptom) with gpiozero 1.6.1-0ubuntu1 and the lgpio backend. I haven't investigated yet, but the debouncing issue in #1039 sounds like a suspect. I'll add some debug prints to check and report back here.

matthijskooijman, 2023-03-22:
Some printing shows that lgpio indeed omits edges, causing the holding status in gpiozero to get stuck (see https://github.com/gpiozero/gpiozero/issues/1039#issuecomment-1479989367 and https://github.com/joan2937/lg/issues/18). The root cause seems to be the kernel not emitting this edge, so I cannot see any fix other than just rechecking the pin whenever you would call `when_held`, as suggested in https://github.com/gpiozero/gpiozero/issues/697#issuecomment-466211367 already.

I've tried this and it seems to work with the following patch against 1.6.1-0ubuntu1 (this might be very much the wrong place to fix this, but it serves to confirm my findings and the direction of such a fix):

```diff
--- mixins.py.orig      2023-03-22 17:30:55.287083624 +0000
+++ mixins.py   2023-03-22 17:30:57.107083625 +0000
@@ -544,6 +544,10 @@
                             ):
                         if parent._held_from is None:
                             parent._held_from = parent.pin_factory.ticks()
+                        # Sometimes edges get missing, so doublecheck if the
+                        # pin is still held and abort if not.
+                        if not parent.value:
+                            break
                         parent._fire_held()
                         if not parent.hold_repeat:
                             break
```

This patch probably leaves the `EventMixin` state incorrect, but a subsequent button down event *is* probably handled and results in when_pressed / when_held as normal (which I had not expected, but I have not investigated how this works exactly).

matthijskooijman, 2023-03-22:
For a more practical workaround (not a fix, obviously) that you can apply in your own program, I have written this `Button` subclass that implements the extra check and if it sees the state is out of sync, it passes the right value to `EventsMixin`:

```python
from gpiozero import Button as ZeroButton

class Button(ZeroButton):
    def _fire_held(self):
        # Sometimes the kernel omits edges, so if the last
        # deactivating edge is omitted held keeps firing. So
        # check the current value and send a fake edge to
        # EventsMixin to stop the held events.
        if self.value:
            super()._fire_held()
        else:
            self._fire_events(self.pin_factory.ticks(), False)
```

lurch, 2023-03-23:
> For a more practical workaround (not a fix, obviously) that you can apply in your own program...

Neat :slightly_smiling_face:   Does this work for both pull-up and pull-down buttons? (it's been a _looong_ time since I've looked at the gpiozero code)

matthijskooijman, 2023-03-23:
I think so - I think the "value" is already normalized to True == active, False == inactive (I have used the above code with an active-low button with a pullup).