gpiozero/gpiozero #699: Add Nokia5110 display support

mancusoa74, 2018-12-09:
Operating system: e.g. Raspbian Stretch
Pi model: e.g. Pi 3 Model B
GPIO Zero version: e.g. 1.4.1
Pin factory used: e.g. RPiGPIO

Hello,
I have written a class to manage the Nokia 5110 display, which is quite a standard small inexpensive display in the maker community.

I would like to contribute to the gpiozero project. 

Would you be interested to add this display to the library?

let me know and I will be happy to issue a PR for the code and documentation.

if yes, against which branch should the PR be issued?

thank you so much
Antonio

bennuttall, 2019-01-08:
Thanks for this - how does the display work? If you can post a [gist](https://gist.github.com/) or something so we can take a look?

SrMouraSilva, 2019-01-15:
@mancusoa74 , @bennuttall, I also implement partially to the PCD8544 (the Nokia 5110 display)

The abstraction idea is based on https://github.com/noxo/SnakePI4J:
* A generic abstract display class with the minimal that displays need to do: color support, size, abstract method to print pixel by pixel
* Displays implementations: https://github.com/PedalPi/Raspberry-Physical/blob/master/component/pcd8544/pcd8544.py
* Drawer class with object forms logic (draw line, draw circle, write text): https://github.com/PedalPi/Raspberry-Physical/blob/master/component/pcd8544/display_graphics.py

As I wrote, the implementation has partially. But is possible understand the idea with the up code.
I have problems in the implementation with the display update speed with the gpiozero 2 years ago. 

I also implemented to Pi4j. In this [repository is the full abstraction](https://github.com/PedalController/Display).

SrMouraSilva, 2019-01-15:
About the speed/slowly, the Linux has (obviously) support to print somethings in monitors with a high speed of updates without a high CPU use with the update frames process.

I researched some time ago and I do not remember for sure. It's possible shows the display on LCD displays (including PCD8544, black white, grayscale displays). There is this library which by what I understand works on Raspberry already https://github.com/notro/fbtft. I do not remember if anything happened to Linux and it stopped working or invented a better protocol and the same author of the code created this library to use the DRM https://github.com/notro/tinydrm/wiki. And also made a pure python version https://github.com/notro/pydrm.

mancusoa74, 2019-01-16:
hi SrMouraSilva,
your job looks very good.

I have no intention to re-implement NOkia5110 display support if it is already exsiting. so please go ahead and I will stop my work.

when it will be merged into master? and hence available to general public?

thnx
Antonio

lurch, 2019-01-16:
Just my 2 cents, but I wonder if graphical displays are "too much" for GpioZero, and would be better off handled by an entirely separate Python library? Feels to me like the concept of a graphical display doesn't really "fit in" at the same level of buttons, leds, etc. that GpioZero works on? :man_shrugging: 

Entirely up to @bennuttall of course though!

bennuttall, 2019-01-16:
Yes I think that's probably true. Another zero-style library for something like that would be good to see though.

mancusoa74, 2019-01-16:
But you can use this type of display also to display string of characters.

so the interface will be very simple and in line with GPIOZEro style.

example:

lcd.write('This is a sample string')

From my point of view, I would love to see GPIOZero as the single library for "Makers"....
Then also supporting chacters LCD like the 16x2 or 20x4, the very common one in the maker community.

:)

lurch, 2019-01-16:
There's no reason that you can't use separate libraries for separate things - Python and pip make this easy. No reason to try and cram everything into a single library :wink:

mancusoa74, 2019-01-16:
of course you can install as many lib as you want.

then following this reasoning, why having Led and Button in GPIOZero. you could have two separated libraries.

in my opinion the point is to have GPIOZero as a "single" overall library for the most common Maker devices. (of course the debate can move now onto the 'most commo')
 :)

lurch, 2019-01-16:
Again it's purely IMHO, but I think the point in GPIOZero is not necessarily "Maker devices", but to make GPIO easy to use. It concerns itself with simple devices that connect "directly" to GPIO pins, e.g. buttons, LEDs, motors, etc. i.e. devices that are only on or off (buttons, LEDs) or devices that have a simple numeric value (motors, PWMLEDs, potentiometers). I think @waveform80 did a really good job with all the sourcetools stuff https://gpiozero.readthedocs.io/en/stable/source_values.html which makes it easy to "plug together" devices, e.g. having a button controlling an LED or having a potentiometer controlling the speed of a motor. Perhaps I'm missing something obvious, but I find it hard to see how something like an LCD screen would fit into that framework, e.g. how would a button or potentiometer affect the 'value' of an LCD screen?

But I suspect we're in danger of going round in circles, so I won't add anything further to this conversation.

mancusoa74, 2019-01-17:
guys, as already said I really believe GPIOZero is very cool, especially for the young maker willing to get started with RPi and typical cheap/school/maker available devices.

For one of my project I need to use Nokia 5110, and I was proposing myself to contribute to this great opensource project.

If the owners/maintainers say no to external contribution for whatever reason, it is part of the game, and we need to respect it. They are the one with big picture in mind. :)

I think the conversation stops here.

I cannot put money into this great project, but I can only put time with contributions. Of course I need to start from angles whic are interesting to me as I am also not paid for it :)

take care