gpiozero/gpiozero #700: Obj must be an instance or subtype of type

filips123, 2018-12-13:
**Environment:**

Operating system: Raspbian Stretch
Pi model: Model B Plus Rev 1.2
GPIO Zero version: 1.4.1
Pin factory used: RPiGPIO

**Description:**

I have some code that is called multiple times per second and sometimes the execution overlaps.

I do not care if sometimes there will be error while getting data, so I made exception handler that simply ignores exception. It often works normally, but sometimes there is an error.

Sometimes the execution is permanently stopped even I handle the exceptions bacause I get error `obj must be an instance or subtype of type`. 

This is weird because it skips the handler. It even skips the try block that was for whole file. Maybe this is problem because I'm using threads and it can't handle it.

Is is possible to fix this?

**Exception:**

```
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gpiozero/pins/rpigpio.py", line 213, in _call_when_changed
    super(RPiGPIOPin, self)._call_when_changed()
TypeError: super(type, obj): obj must be an instance or subtype of type
```

lurch, 2018-12-24:
I think it'll be hard for the devs to know what's going on unless you post some example code?

filips123, 2018-12-24:
I don't know what exactly was the code, because I replace it with working one, but it was something like this:

```python
import socketserver

class ThreadingUnixServer(socketserver.ThreadingMixIn, socketserver.UnixStreamServer):
    pass

import json
from gpiozero import DistanceSensor, LineSensor

def example(server):
    ultrasonic = DistanceSensor(echo=17, trigger=4, max_distance=5)
    left = LineSensor(27)
    right = LineSensor(22)

    response = {
        'distance': ultrasonic.distance * 100,
        'line-left': -left.value + 1,
        'line-right': -right.value + 1,
    }
    server.wfile.write(bytes(json.dumps(response) + '\n', 'utf-8'))

# This function is than called from handler for threading unix server
# Handler is passed as function argument
```

lurch, 2018-12-24:
> I have some code that is called multiple times per second and sometimes the execution overlaps.

`DistanceSensor` is timing based (it has to send a ping and wait for the reply), so I think if you want to call it from a thread you'd need to use some kind of mutex or lock to ensure that it doesn't get called multiple "overlapping" times.

filips123, 2018-12-24:
And why `try ... except` blocks didn't catch this? Is this because of threads?
Anyway, I fixed this and I now only want to know, why this happened.

lurch, 2018-12-24:
Without being able to run your non-working code, it'll probably be impossible to say for sure what was going wrong?

filips123, 2018-12-24:
I can't provide you that code and I have already fixed this. Handler probably didn't worked because of threads?

It can be fixed using global pre-initialization of sensors. I add them to handler class and then pass it to function. It now works.

This can be closed.

closed 2018-12-24 by filips123