gpiozero/gpiozero #701: When exit program, it show error message with self design package.

lagagain, 2018-12-14:
Operating system: e.g. Raspbian Stretch
Pi model: e.g. Pi 3 Model B
GPIO Zero version: e.g. 1.4.1
Pin factory used: e.g. RPiGPIO
See http://rpf.io/gpzissue for information on how to find out these details
Python version: 2.7.13 (but I think Python3 have same problem)

I make a package for myself. When I import it and exit program, it shows some error message like below:

```shell
Python 2.7.13 (default, Sep 26 2018, 18:42:22) 
[GCC 6.3.0 20170516] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import device
>>> exit()
Exception TypeError: 'super() argument 1 must be type, not None' in <object repr() failed> ignored
Exception TypeError: 'super() argument 1 must be type, not None' in <object repr() failed> ignored
Exception TypeError: 'super() argument 1 must be type, not None' in <object repr() failed> ignored
Exception TypeError: 'super() argument 1 must be type, not None' in <object repr() failed> ignored
Exception TypeError: TypeError('super() argument 1 must be type, not None',) in <object repr() failed> ignored
Exception TypeError: TypeError('super() argument 1 must be type, not None',) in <object repr() failed> ignored
Exception TypeError: TypeError('super() argument 1 must be type, not None',) in <object repr() failed> ignored
Exception TypeError: 'super() argument 1 must be type, not None' in <object repr() failed> ignored
Exception TypeError: 'super() argument 1 must be type, not None' in <object repr() failed> ignored
Exception TypeError: 'super() argument 1 must be type, not None' in <object repr() failed> ignored
Exception TypeError: TypeError('super() argument 1 must be type, not None',) in <object repr() failed> ignored
Exception TypeError: TypeError('super() argument 1 must be type, not None',) in <object repr() failed> ignored
Exception TypeError: TypeError('super() argument 1 must be type, not None',) in <object repr() failed> ignored
```

-------

device package struct:
```
|-- device
|   |-- device.py
|   |-- device.pyc
|   |-- __init__.py
|   |-- __init__.pyc
|   |-- pin.py
|   `-- pin.pyc
```
pin.py just some config for setting PIN number.  And device.py original code show below:

```python
import pin
from gpiozero import Motor
from gpiozero import LineSensor
from gpiozero import DistanceSensor

Motor_Right = Motor(pin.Motor_Right.FORWARD_PIN,
                    pin.Motor_Right.BACKWARD_PIN)

Motor_Left = Motor(pin.Motor_Left.FORWARD_PIN,
                   pin.Motor_Left.BACKWARD_PIN)

Line_Left = LineSensor(pin.Trace_Left.TRACE_PIN)
Line_Right = LineSensor(pin.Trace_Right.TRACE_PIN)
Line_Middle = LineSensor(pin.Trace_Middle.TRACE_PIN)

Distance_Front = DistanceSensor(pin.Distance_Front.ECHO_PIN,
                                pin.Distance_Front.TRIGGER_PIN)
Distance_Back = DistanceSensor(pin.Distance_Back.ECHO_PIN,
                               pin.Distance_Back.TRIGGER_PIN)
Distance_Left = DistanceSensor(pin.Distance_Left.ECHO_PIN,
                               pin.Distance_Left.TRIGGER_PIN)
Distance_Right = DistanceSensor(pin.Distance_Right.ECHO_PIN,
                                pin.Distance_Right.TRIGGER_PIN)
```

-------------------

It just happen at exiting program, and not a big problem.  But still are some warning to notice and fix.

lagagain, 2018-12-14:
change to under:

```python
import pin
from gpiozero import Motor
from gpiozero import LineSensor
from gpiozero import DistanceSensor

class Motor_Right(Motor):
    def __init__(self):
        super(Motor_Right, self).__init__(pin.Motor_Right.FORWARD_PIN,
                       pin.Motor_Right.BACKWARD_PIN)

class Motor_Left(Motor):
    def __init__(self):
        super(Motor_Left, self).__init__(pin.Motor_Left.FORWARD_PIN,
                       pin.Motor_Left.BACKWARD_PIN)

class Line_Left(LineSensor):
    def __init__(self):
        super(Line_Left, self).__init__(pin.Trace_Left.TRACE_PIN)
class Line_Right(LineSensor):
    def __init__(self):
        super(Line_Right, self).__init__(pin.Trace_Right.TRACE_PIN)
class Line_Middle():
    def __init__(self):
        super(Line_Middle, self).__init__(pin.Trace_Middle.TRACE_PIN)

class Distance_Front():
    def __init__(self):
        super(Distance_Front, self).__init__(pin.Distance_Front.ECHO_PIN,
                                             pin.Distance_Front.TRIGGER_PIN)
class Distance_Back():
    def __init__(self):
        super(Distance_Back, self).__init__(pin.Distance_Back.ECHO_PIN,
                                             pin.Distance_Back.TRIGGER_PIN)
class Distance_LeftI():
    def __init__(self):
        super(Distance_LeftI, self).__init__(pin.Distance_Left.ECHO_PIN,
                                             pin.Distance_Left.TRIGGER_PIN)
class Distance_Right():
    def __init__(self):
        super(Distance_Right, self).__init__(pin.Distance_Right.ECHO_PIN,
                                             pin.Distance_Right.TRIGGER_PIN)
```
and use below:
```
bpython version 0.16 on top of Python 2.7.13 /usr/bin/python
>>> import device
>>> motor_left = device.Motor_Left()
>>> motor_left.forward()
>>> motor_left.stop()
>>> exit()
(None,)
```

Will not happen error like before.

lurch, 2018-12-24:
Could you explain what it is you're trying to do by making your own package?
If you're trying to make use of GpioZero then I think you can't have files named `device.py` or `pin.py` as these conflict with files of the same name used by GpioZero... (or perhaps that's only true for packages and not modules?) :man_shrugging:

lurch, 2018-12-24:
> who I need to @ to close this issue?

I think you should be able to close the issue yourself? (as you're the person who opened it)

lagagain, 2018-12-24:
It happen because name conflict.

Close this issue

closed 2018-12-24 by lagagain