gpiozero/gpiozero #702: An attempt of expanding to circuitpython

ezeakis, 2018-12-22:
It is not an actual problem, but I did not know in what other way to refer this to you.

I am using gpiozero with rpi projects and it is great. I wanted to introduce my ninjas in my dojo to microcontrollers but not make them learn a new language, so I chose Adafruit Trinket M0 and created a minimalized module that it would work like gpiozero for a LED but I named it mygpiozero in order not to confuse anyone that it would not be the original. Well, it worked like a charm. The kids programmed the controller just like the RPi.

I think it would be a nice idea to make these two worlds meet. If you want some more information you may find it here.

https://coyote.edu.gr/en/introduction_to_trinket_m0-2/

bennuttall, 2019-01-08:
That's cool. I think only the basic functionality of gpiozero can be made available in {micro,circuit}python, as things like threaded callbacks and [source/values](https://gpiozero.readthedocs.io/en/stable/source_values.html) (again, threaded) wouldn't work within the memory constraints, but the device class, DSL and Pythonic nature of gpiozero would certainly be better suited than the low-level standard method.

I don't really know where any effort to create gpiozero in the micro world would/should live.

Ping @ladyada

lurch, 2019-01-08:
On a side-note, the Python code demonstrated in https://www.raspberrypi.org/blog/making-robot-friends-with-the-crickit-hat-for-raspberry-pi/ reminded me very much of GpioZero :wink: 
I dunno if that was deliberate, or if both projects happened to independently develop similar "pythonic" APIs. :snake:

ladyada, 2019-01-08:
hiya yah circuitpython's hardware api was heavily influenced by gpiozero, which you can see especially when working with expanders and ADCs - but gpiozero was not complete enough for all the stuff we wanted to do like I2C/SPI: 
https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/circuitpython-raspi#wait-isnt-there-already-something-that-does-this-gpio-zero-2-3

we have `adafruit-blinka` - our compatibility library for linux computers, so that they can use our libraries. and it works pretty well! it might be easier for the kids to program both using the circuitpython api? (see guide above) - there's a lot of stuff in gpiozero that will never fit on a trinket like the interrupt stuff :) 

a gpiozero compatibility class for circuitpython could be handy for people as long as they're aware its minimal - if you want to have `gpiozero` added to our bundle of circuitpy libraries, check out this guide on how to put it onto github https://learn.adafruit.com/creating-and-sharing-a-circuitpython-library/ we're also on discord and in the `adafruit/circuitpython` repo if you need a hand :)

ladyada, 2019-01-08:
(just as an aside, we use libgpiod, smbus and spidev to support any/all linux SBCs without needing mmio specialities)

bennuttall, 2019-01-08:
I think it makes sense to keep learning from each other as we move forward, but it's infeasible to share a codebase or anything like that.

> https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/circuitpython-raspi#wait-isnt-there-already-something-that-does-this-gpio-zero-2-3

Ooh I hadn't seen that before. Thanks for the shout-out :+1: 

> (just as an aside, we use libgpiod, smbus and spidev to support any/all linux SBCs without needing mmio specialities)

I need to look at libgpiod. It could be the basis of a new pin factory for us. We use spidev (and have a GPIO-based software fallback) and might use smbus for I2C when we add that (planned).

closed 2019-01-08 by bennuttall

ezeakis, 2019-01-08:
Thank you very much for your time checking this attempt/thought and your comments! Certainly, this was a minimalized version just for this LED project, but I have some things in mind for future dojo sessions and it might fit in. I will try both to extend this attempt and the other proposals as well. Thank you again very much!

bennuttall, 2019-01-08:
@ladyada Also re: the [next page](https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/installing-circuitpython-on-raspberry-pi) of your guide which includes installing libraries with pip, here are the top Adafruit packages on piwheels:

```
                package                 | downloads
----------------------------------------+----------
 Adafruit-PureIO                        |     22196
 Adafruit-GPIO                          |     20675
 Adafruit-Blinka                        |     10837
 adafruit-circuitpython-busdevice       |      6053
 Adafruit-DHT                           |      6032
 Adafruit-PCA9685                       |      3245
 Adafruit-ADS1x15                       |      2226
 adafruit-circuitpython-neopixel        |      2107
 adafruit-circuitpython-register        |      1458
 Adafruit-PlatformDetect                |      1450
 Adafruit-MCP3008                       |      1293
 adafruit-io                            |      1286
```

That places `Adafruit-PureIO` at `#53` overall!

ladyada, 2019-01-08:
not surpsing, we depend on it a lot :) how did you get that output?

dglaude, 2019-01-08:
I don't know if it make a lot of sense, but maybe GpioZero could use Blinka as it's pin factory.
That would instantly make it portable on every linux place where Blinka runs.

It might not be in the best short term interest for the Raspberry Pi Foundation, however since they will be more or less force to move to some other CPU for the next Pi generation, the sooner they break away from BCM specific, the easier it will be.

ladyada, 2019-01-09:
ben, is there any way i could see those numbers myself or do ya have to perform a query - i'd be curious to see how this changes as we move more towards Blinka and away from Adafruit-GPIO! :)

bennuttall, 2019-01-10:
They're not publicly available (we might be able to make them available on project pages in near future). I am just running queries but I don't mind sending you some results - can you email me? ben@

ladyada, 2019-01-10:
ben, dya know if these stats are included in the pypi stats? if so, we could query those via the api instead of having you run a query by hand!

bennuttall, 2019-01-10:
piwheels downloads won't be included in pypi stats, no.

ladyada, 2019-01-10:
oooh ok - thats interesting but makes sense. ok let me get back to you!

bennuttall, 2019-01-10:
I have a graph for you :grinning:

ladyada, 2019-01-10:
ok ok i'll email :D

lurch, 2019-06-19:
> I need to look at libgpiod.

Just out of curiosity, I was looking at that this evening. It supports reading and writing GPIO values, and monitoring for events, but unfortunately it doesn't support setting pull-ups or pull-downs :confused: 
More info at https://github.com/brgl/libgpiod/issues/31