gpiozero/gpiozero #703: still getting PinFactoryFallback warnings after installing rpi.gpio

Pitichinaccio, 2018-12-22:
I'm still struggling with PinFactoryFallback warnings, even after installation of rpi.gpio.

Operating system: Raspbian Stretch
Pi model: Pi 3 Model B
GPIO Zero version: 1.4.1

I always get this error messages:
```
Dec 22 18:17:17 raspberrypi snips-skill-server[500]: INFO:snips_skill_server_lib::runner            : [setGPIO][err] /var/lib/snips/skills/snips-setGPIO/venv/local/lib/python2.7/site-packages/gpiozero/devices.py:452: PinFactoryFallback: Falling back from rpigpio: No module named RPi
Dec 22 18:17:17 raspberrypi snips-skill-server[500]: INFO:snips_skill_server_lib::runner            : [setGPIO][err]   'Falling back from %s: %s' % (name, str(e))))
Dec 22 18:17:17 raspberrypi snips-skill-server[500]: INFO:snips_skill_server_lib::runner            : [setGPIO][err] /var/lib/snips/skills/snips-setGPIO/venv/local/lib/python2.7/site-packages/gpiozero/devices.py:452: PinFactoryFallback: Falling back from rpio: No module named RPIO
Dec 22 18:17:17 raspberrypi snips-skill-server[500]: INFO:snips_skill_server_lib::runner            : [setGPIO][err]   'Falling back from %s: %s' % (name, str(e))))
Dec 22 18:17:17 raspberrypi snips-skill-server[500]: INFO:snips_skill_server_lib::runner            : [setGPIO][err] /var/lib/snips/skills/snips-setGPIO/venv/local/lib/python2.7/site-packages/gpiozero/devices.py:452: PinFactoryFallback: Falling back from pigpio: No module named pigpio
Dec 22 18:17:17 raspberrypi snips-skill-server[500]: INFO:snips_skill_server_lib::runner            : [setGPIO][err]   'Falling back from %s: %s' % (name, str(e))))
Dec 22 18:17:17 raspberrypi snips-skill-server[500]: INFO:snips_skill_server_lib::runner            : [setGPIO][err] /var/lib/snips/skills/snips-setGPIO/venv/local/lib/python2.7/site-packages/gpiozero/devices.py:452: PinFactoryFallback: Falling back from native: unable to open /dev/gpiomem or /dev/mem; upgrade your kernel or run as root
Dec 22 18:17:17 raspberrypi snips-skill-server[500]: INFO:snips_skill_server_lib::runner            : [setGPIO][err]   'Falling back from %s: %s' % (name, str(e))))
Dec 22 18:17:17 raspberrypi snips-skill-server[500]: INFO:snips_skill_server_lib::runner            : [setGPIO][err] Traceback (most recent call last):
Dec 22 18:17:17 raspberrypi snips-skill-server[500]: INFO:snips_skill_server_lib::runner            : [setGPIO][err]   File "./action-setGPIO.py", line 9, in <module>
Dec 22 18:17:17 raspberrypi snips-skill-server[500]: INFO:snips_skill_server_lib::runner            : [setGPIO][err]     from gpiozero import LED
Dec 22 18:17:17 raspberrypi snips-skill-server[500]: INFO:snips_skill_server_lib::runner            : [setGPIO][err]   File "/var/lib/snips/skills/snips-setGPIO/venv/local/lib/python2.7/site-packages/gpiozero/__init__.py", line 22, in <module>
Dec 22 18:17:17 raspberrypi snips-skill-server[500]: INFO:snips_skill_server_lib::runner            : [setGPIO][err]     from .devices import (
Dec 22 18:17:17 raspberrypi snips-skill-server[500]: INFO:snips_skill_server_lib::runner            : [setGPIO][err]   File "/var/lib/snips/skills/snips-setGPIO/venv/local/lib/python2.7/site-packages/gpiozero/devices.py", line 486, in <module>
Dec 22 18:17:17 raspberrypi snips-skill-server[500]: INFO:snips_skill_server_lib::runner            : [setGPIO][err]     Device.pin_factory = _default_pin_factory()
Dec 22 18:17:17 raspberrypi snips-skill-server[500]: INFO:snips_skill_server_lib::runner            : [setGPIO][err]   File "/var/lib/snips/skills/snips-setGPIO/venv/local/lib/python2.7/site-packages/gpiozero/devices.py", line 453, in _default_pin_factory
Dec 22 18:17:17 raspberrypi snips-skill-server[500]: INFO:snips_skill_server_lib::runner            : [setGPIO][err]     raise BadPinFactory('Unable to load any default pin factory!')
Dec 22 18:17:17 raspberrypi snips-skill-server[500]: INFO:snips_skill_server_lib::runner            : [setGPIO][err] gpiozero.exc.BadPinFactory: Unable to load any default pin factory!
Dec 22 18:17:17 raspberrypi snips-skill-server[500]: WARN:snips_skill_server_lib::handler::nosandbox: child setGPIO has stopped with status `exit code: 1`
```
I did install rpi.giozero using
`
pip install rpi.gpio`

but nothing changed.
Any ideas how to solve this problem? Many thanks!

bertron

bennuttall, 2019-01-08:
What Python version? If you're using Python 3, make sure you install it with `pip3` not `pip`.

You can test you have RPi.GPIO installed by running a `python3` prompt and importing it.

bennuttall, 2019-01-08:
Oh I see you're in a Python 2 virtualenv. I guess just make sure you have RPi.GPIO installed correctly. If it can't import it normally, either it's not installed or there's an issue importing it, like a name clash.

It's possible that if you have a file named `RPi.py` or similar the RPi.GPIO import line will try to import your file not the library. Test this import alone and see if the error message gives you any new information.

closed 2019-02-01 by bennuttall