gpiozero/gpiozero #708: Allow users to differentiate between pressed and held events

bennuttall, 2019-01-15:
I have written an [FAQ](https://gpiozero.readthedocs.io/en/latest/faq.html#how-do-i-use-button-when-pressed-and-button-when-held-together) on this but @waveform80 thinks he can do it another way.

lurch, 2019-01-15:
When looking at #685 I found myself wondering if it might be useful for the `when_released` callback to be able to determine how long the button was held down for before being released? That way you could have different-length button presses triggering different actions, without having to rely on `when_held`.
I had a look at the `EventsMixin` and I think adding a `_last_state_duration` attribute would allow that? Maybe I should put this in a new issue...?

kenomaerz, 2020-03-22:
+1 
This is a more or less regular usecase, I do believe.